vilemduha/blendercam #5: Imperial unit export broken

dastultz, 2015-10-10:
`utils.py` has 2 places that say `'INCH'`. They need to be changed to `'IMPERIAL'` to produce the expected output.

Thanks

vilemnovak, 2015-10-10:
thanks for noting this, I fixed this and commited. i am quite curious when
this actually got in the repository, have to look at that...

dastultz, 2015-10-11:
Looks like Denver Hull supplied the patch for this. Not sure what he intended to fix, I hope this doesn't break it...

Did you push? I don't really need you to push, I'm just not all that well versed in Git (I use Mercurial daily) so I'm not sure if I'm doing something wrong.

vilemnovak, 2015-10-11:
Hi, maybe you'll consider that strange, but I don't precisely know what is it to push  in git :) I call it probably commit, maybe because I used svn before...

dastultz, 2015-10-11:
Yes, no need for us to squabble over differences in source control systems. My point is that when I go here: https://github.com/vilemnovak/blendercam/commits/master I don't see the commit you are referring to. Like I said, it's not holding me back, but when I see the fix in place I'll close the issue. Thanks!

nfz, 2015-12-14:
This seems to be fixed.  I switched to imperial and the gcode exported is indeed in imperial when I select imperial in Blendercam.

The commit was here:
https://github.com/vilemnovak/blendercam/commit/64fc0cd8a5190958a4a8fd8bd784b7dacc2d3c14

vilemnovak, 2015-12-15:
indeed, closed it now

closed 2015-12-15 by vilemnovak