vilemduha/blendercam #8: stuck in sorting path

ecl66, 2015-12-18:
first of all, thank you so much for this wonderful work you are doing with blendercam!

after using the latest commit, it gets stuck in  sorting paths (parallel strategy) - I am using blender 2.70 (under ubuntu 15.10). any ideas where this comes from?

all the best and thank you again for providing what is probably the best CAM option for linux so far!

nfz, 2015-12-18:
Yup, getting the same thing.  I have a fix for it in my fork of Blendercam and will generate a pull request soon.

vilemnovak, 2015-12-18:
Thanks for a very nice feedback - 
the last commit is largely unfinished, I wanted to commit it at least in this state. P
nfz - if you have a fix, please upload it, or if you believe it

nfz, 2015-12-24:
I am not getting the infinite loop in sortChunks() with with the most recent updates in Master.

ecl66, 2015-12-25:
Hi Guy, I'm still getting this problem using the latest update (checked out today) and Blender 2.70.
Here is the output of the program, cutting a 80x120x30mm object with a 3mm end cutter:

Operation_1 sampling paths  93.0%
Operation_1 sampling paths  94.0%
Operation_1 sampling paths  95.0%
Operation_1 sampling paths  96.0%
Operation_1 sampling paths  97.0%
Operation_1 sampling paths  98.0%
Operation_1 sampling paths  99.0%
Operation_1 checking relations between paths
Operation_1 sorting paths

How long should it take (I've been waiting for 30 minutes now, on a very fast computer (Intel® Core™ i7-4710HQ CPU @ 2.50GHz × 8 , 16Gb RAM on Ubuntu 15.10 and Samsung SSD)?

ecl66, 2015-12-25:
Hello again,

Just for info, my model has about 5540 chunks.

Sorting issue seems to only occurs when calculating path in background, but not when using the normal calculate mode...

Hope that helps!

ecl66, 2015-12-25:
Also occurs in normal calculate mode (i.e. not background) when using layers in parallel strategy...

nfz, 2015-12-26:
30 minutes is way too long.  Definitely a problem.  Do you have a .blend that I could test?  The other thing you could try is to uncomment(remove the leading #) the following lines in utils.py:
1576
1583
1584
1585
Make sure you use the latest from github.

ecl66, 2015-12-26:
Hi Jeff,

Here is the .blend.

I am using the lastes from github and indeed already tried to comment 
those lines in utils.py

Cheers,
Emmanuel

nfz, 2015-12-30:
The .blend file does not seem to be attached to your last comment.

nfz, 2015-12-31:
Try the latest, there was a fix for sorting chunks.

vilemnovak, 2016-01-16:
this one was obviously fixed, closing.

closed 2016-01-16 by vilemnovak