vilemduha/blendercam #9: Enable Wikis

nfz, 2015-12-18:
It would be nice if the Wikis feature for this project was enabled. I would like to start contributing to a user manual of some sort and the Wikis would be a good place to start.

vilemnovak, 2015-12-18:
Hi, a manual is started on the Blender CAM page - can you maybe continue
there? Or do you think wiki is better?
<https://www.avast.com/?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Tento
email byl odeslán z počítače bez virů, chráněného programem Avast.
www.avast.com
<https://www.avast.com/?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

nfz, 2015-12-19:
The wiki would allow anyone to add/edit the manual.

vilemnovak, 2015-12-19:
Ok, enabled wikis now!

<https://www.avast.com/?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Tento
email byl odeslán z počítače bez virů, chráněného programem Avast.
www.avast.com
<https://www.avast.com/?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2015-12-20 by nfz