vilemduha/blendercam #14: Bump version number of Blendercam

nfz, 2015-12-24:
Can we bump the version number of the addon? Maybe to "0.8.1" or "0.9.0" since there have been a lot of changes, bugfixes, and improvements.  Would make it easier to differentiate who is using Blendercam Github version or another one when discussing problems on the forum.

vilemnovak, 2016-01-13:
I would like to release as soon as possible - I always do repack of the binaries on my website. It's important that people keep visiting the original site, because that's the only way to get donations now, and most people want to install a version "out of the box".
So, we can mark a certain revision as a release, then the binaries can be provided. If there are no crucial bugs, we could do it now...

vilemnovak, 2016-01-17:
Ok, so I am all for a release, I will update the widnows binaries, and try to upload as soon as possible. I will also try to contact others for Mac and Linux.

closed 2016-01-31 by vilemnovak