vilemduha/blendercam #15: Wiki - Using Blendercam from github

narvf, 2016-01-08:
Missin informations what librariers should be installed first, at least on windows.
And what blender version is the best to use.
[image]

nfz, 2016-01-09:
Sorry,  I have not finished that section.  Didn't think anyone would look at it yet :).
The only dependency you need to install now is Shapely.

I build blender daily and use that.  Some people also use the daily build which amounts to the same thing.  The latest blender official release works fine too.

narvf, 2016-01-12:
Is possible to add a little guide about installing shapely?
pip install shapely  with libgeos installed seems not to work on my home pc with linux mint.
I just downloaded zip archiwe from github and extracted into fresh blender directory and still not have blendercam available.

ksuprynowicz, 2016-01-14:
Hi Narvf,
I don't know if it'll work on Linux Mint, but on Debian Testing it was enough to install packages:

    python3-shapely python3-numpy blender

Then start Blender from the system path(not one downloaded separately) and add BlenderCAM scripts directory in blender preferences. As Mint uses 'apt' package manager it should be the same. There was no need to use 'pip' package manager at all.

narvf, 2016-01-14:
Thanks! Now I can try some of new features.

closed 2016-01-15 by narvf