vilemduha/blendercam #16: Meaning: when you change the way the tasks workspace the patch is shifted relative to the model

torvn77, 2016-01-09:
Sorry, but I don't know English, this will simply leave the link to the video:
https://youtu.be/MNBpCfhOeGo

Meaning: when you change the way the tasks workspace the patch is shifted relative to the model

Original text ( Russian ) -----------------------------------------------------
Извините, но я не знаю английский, по этому просто оставлю ссылку на видео:
https://youtu.be/MNBpCfhOeGo

Смысл: при изменений способа задачи рабочей области патч смещается относительно

evgeniident, 2016-01-09:
I also have the problem that the offset as seen in the video: https://youtu.be/MNBpCfhOeGo
These things I have observed in other strategy, partly even with a new calculation without! changes

torvn77, 2016-01-10:
Essence in that if to remove a Daw "Estimate from model", and set the material in the manual, the patch is correctly calculated.
If this Daw to remove and set the material through of how it needs to be more models, then the patch is evaluated at an offset from the model
in any case, if the processing area is limited by the circuit.
The presence of this bug in the absence of limiting a processing area of the circuit I haven't checked.
As it looks like you can see on the video.
https://youtu.be/MNBpCfhOeGo

Original text (Russian):
Сущность в том, что если снять галку "Estimate from model" и задать материал в ручную, то патч вычисляется правильно.
Если же эту галку снять и задать материал через то, насколько он должен быть больше модели, то патч вычисляется со смещением относительно модели,
во всяком случае, если область обработки ограничена контуром.
Наличие этого бага при отсутствии ограничивающего область обработки контура я не проверял.
Как это выглядит можно посмотреть на ролике.

vilemnovak, 2016-01-13:
I have to say, I don't understand the issue precisely, from the explanations, and not from the video..please upload  a file.

torvn77, 2016-01-14:
The previous file *,the mixture is wiped off and to re-reproduce the bug comes out.
Don't know what it is, maybe in the last update, what can button is not pressed.
And the moment on video, which shows the bug actually 5 seconds play video
https://youtu.be/MNBpCfhOeGo?t=5s

vilemnovak, 2016-01-17:
Great, I will close for now, feel free to reopen the issue if it returns ;)

closed 2016-01-17 by vilemnovak