vilemduha/blendercam #17: Wrong gcode axis rotation when using 4th axis indexing

nfz, 2016-01-09:
See discussion on google group:
https://groups.google.com/forum/embed/?place=forum%2Fblender-cam-forum&theme=softgray&hideforumtitle=true&showsearch=false&showpopout=true&showtabs=true&parenturl=http%3A%2F%2Fblendercam.blogspot.ca%2Fp%2Fblog-page.html&afterlogin=&pli=1#!category-topic/blender-cam-forum/development/hQCkP0E5-g8

The A axis should be used for 4th axis indexing?  I have built my own 4th axis along with controller and I have that 4th axis as A based on the literature I have read on the Internet.  It should not matter if the 4th axis is orientated x,y, or z.  The g-code that Blendercam generates uses B axis for rotated indexing when the 4th axis is X.  If I switch the return values in rotTo2axes() in utils.py then A axis gets used properly.

Things get stranger when the 4th axis is selected to Y or Z but I will expand on that later.

vilemnovak, 2016-01-09:
This is totally unfinished part of Blender CAM - 4 axis code works and was tested, but the machine configuration should be part of the machine settings panel. It was meant to be that way, and I think there are the options for these settings, but not sure how far I got with this back when I was working on this part.

nfz, 2016-01-09:
Ok, will dig deeper.

nfz, 2016-01-09:
I found the commented out code that partially deals with this problem.   The Machine Settings properties axis4 and axis5 where commented out in ui.py.  They are not used in utils.py so I will cobble something together as I need indexing working today for a project.  Will be a good way to learn 4th axis stuff.

nfz, 2016-01-22:
I got it working for the project I needed to finish but I will hold off on a pull request until I get the NC update done.

vilemnovak, 2016-02-06:
what about this? maybe might go on now?

nfz, 2016-02-10:
Yep, will commit.

closed 2023-01-14 by pppalain