vilemduha/blendercam #18: Button positioning model

torvn77, 2016-01-10:
[image]
Hello, I want to offer you in the section "CAM Material size and Position" button instead of the "Position Object" to make two buttons:
1)the First button "Zero of material surface" will configure BlenderCAM for creating NC programs for engraving,
the object will be positioned at the top of the lower region Z<0, and the quick movements will be above zero.
[image]

2)the Second button "Zero of W
[image]
orktable" button will netravati Blendercam to create NC programs for 3D milling reliefs.
The zero pack is placed on the surface of the desktop, and the whole process is conducted in the positive part Z>0
[image]

I certainly poorly explained, to understand the character of the query show the text and illustrations attached to the milling professional and he will explain what it is.

Original text:
Здравствуйте, хочу предложить вам в секции "CAM Material size and Position" вместо кнопки "Position Object" сделать две кнопки:
1)Первая кнопка "Zero of material surface" будет настраивать BlenderCAM для создания УП для гравировки,
при этом объект будет позиционироваться в верху нижней области Z<0, а область быстрых перемещений будет выше нуля.
2)Вторая кнопка "Zero of Worktable" кнопка будет настравать Blendercam на создание УП для 3D фрезерования рельефов.
При этом ноль УП распологается на поверхности рабочего стола, а весь процесс ведётся в положительной части Z>0

Я конечно плохо изложил, чтобы понять сущьность запроса покажите этот текст и приложенные иллюстрации профессиональному фрезеровщику и он вам объяснит в чём дело.

nfz, 2016-01-22:
Ok, I could see this being useful.

nfz, 2016-01-22:
I will put together a little test.

nfz, 2016-01-22:
Something like this?
[image]

torvn77, 2016-01-24:
>Something like this?
Yes.

closed 2023-01-14 by pppalain