vilemduha/blendercam #19: Performance of sortChunks with multilayer Parallel operation

ksuprynowicz, 2016-01-13:
For multiple-layer Parallel operation sortChunks function makes path calculation very slow.
Included paralllel operation computes in 7 second with single layer and 830 seconds with 3 layers. For more complex operations it takes even longer.

According to cProfile Python built-in profiler, 99% of calculation time for 3-layer operation is spent in getNextClosest function called from sortChunks( in total 15740 calls, 0.049s each):

    ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.000    0.000  829.630  829.630 /home/ksuprynowicz/blender_cam_SE_mk3/blendercam/scripts/addons/cam/ops.py:195(execute)
        1    0.006    0.006  829.630  829.630 /home/ksuprynowicz/blender_cam_SE_mk3/blendercam/scripts/addons/cam/utils.py:3181(getPath)
        1    0.003    0.003  826.420  826.420 /home/ksuprynowicz/blender_cam_SE_mk3/blendercam/scripts/addons/cam/utils.py:2323(getPath3axis)
        1    0.032    0.032  821.230  821.230 /home/ksuprynowicz/blender_cam_SE_mk3/blendercam/scripts/addons/cam/utils.py:1470(sortChunks)
    15740  765.601    0.049  821.094    0.052 /home/ksuprynowicz/blender_cam_SE_mk3/blendercam/scripts/addons/cam/chunk.py:132(getNextClosest)

For single layer there are only 68 calls, each less than 1 millisecond:
       68    0.000    0.000    0.000    0.000 /home/ksuprynowicz/blender_cam_SE_mk3/blendercam/scripts/addons/cam/chunk.py:132(getNextClosest)

In earlier versions(probably pre-2015) there was no such problem.

Link to the test file with single and multilayer operation used for profiling:
https://drive.google.com/file/d/0B0C06H4DvveCX2dXbUEwVEExRGc/view?usp=sharing

vilemnovak, 2016-01-13:
Great profiling Karol... what tool are you using to do such profiling? As you know, It's rather a miracle I developed Blender CAM, not knowing much about programming. IT's definitely worth looking into this. 
The thing is - I assume pre-2015 versions didn't have good sorting at all...
However, getNextClosest seems to be looping a bit too much here, there is definitely a huge space for optimization.

ksuprynowicz, 2016-01-13:
Thanks!

Sadly, I'm a bit stuck at figuring why the slowdown happens...

Python has a builtin profiler, to call it you have to:

    import cProfile

and then to run the operation:

    cProfile.run("bpy.ops.object.calculate_cam_path()", "op.prof")

The results will be written to "op.prof" file. To view the results:

    import pstats
    stats = pstats.Stats('op.prof')
    stats.sort_stats('cumulative')
    stats.reverse_order()
    stats.print_stats()

"stats.reverse_order()" is there so the most time-intensive functions will show up last in the Python console

vilemnovak, 2016-01-14:
Thanks Karol, I've learned something new and before all, this 'bug' is fixed now ;) I think it could also optimize sorthing of other operations.

closed 2016-01-14 by vilemnovak

ksuprynowicz, 2016-01-14:
Thanks! Now instead of 830 seconds its 8 seconds - 100 times faster :) As I machine a lot of aluminum alloy parts, this feature is really important to me.

vilemnovak, 2016-01-14:
I understand, I usually do rough milling with few layers(2-3 max), where the issue wasn't so bad... with more layers the problem was really terrible.