vilemduha/blendercam #20: Overcut option?

narvf, 2016-01-14:
Is it possible to add an overcut option in corners? I have no experience in programming. 
This is useful for example in plywood when i have two or more elements to put on inside another.

Like here for example
http://www.cambam.info/doc/plus/images/overcut.png

vilemnovak, 2016-01-14:
Since there are so many overcut options, I so far decided not to try to add this feature. It would have to have manual input anyway probably...

nfz, 2016-01-14:
I do it manually using the drill strategy as an operation.  I just mark all the corners that need overcut and do it as a chained operation.  I use a circle curve the diameter of the bit (usually an end mill) and then its easy to place exactly on the corner.  Just use alt-D to clone the curve while in edit mode for each corner and move the curve on the z axis to the depth of cut needed.  I guess if you have a lot of corners to do then automating it would be faster.

vilemnovak, 2016-01-14:
Well it would, especially also because the overcut is correct when you do not put a circle in the corner, you just need to touch the corner with the side of the circle. I always thought about overcuts as part of design, not relaly something CAM should do...

narvf, 2016-01-14:
Great idea nfz. I do my projects in inkscape and the import svg filem. Se it's not such a big problem to add circles and use drill option :)

vilemnovak, 2016-01-14:
Hi, just commited overcuts operator.
Select curve or text object - hit space, write overcut.

Let me know if this works correctly.

nfz, 2016-01-15:
Checking it out...

nfz, 2016-01-15:
I must be missing something.  I set up a square using a bezier circle and applied the overcuts operator.  A curve is created with the identical shape (a square).  I played with the diamter and threshold and can see in the terminal window new calculations being done but no change in the shape.
[image]

[blendercamovercut.blend.zip](https://github.com/vilemnovak/blendercam/files/91374/blendercamovercut.blend.zip)

I will keep playing and take a look at the code.  Probably silly user error.

vilemnovak, 2016-01-15:
just check also outer polygons option. - it is quite logical to make the overcuts only in holes, and your shape doesn't have any hole - the algorithm assumes it is working on  a whole shape which is to be cut. If you check the

vilemnovak, 2016-01-15:
hm ok - so the algo is written this way - it adds overcuts only to inner angles. I will add an invert option, where it's assumed that you actually want to create only the hole.
Test it with some text object to see how it works now..

vilemnovak, 2016-01-15:
also, remember you can always use the Intarsion operator, which is there to make perfect inlays - shapes that can be milled both from inside and from outside.

vilemnovak, 2016-01-15:
changed this and now things should give more sense. 
here you can see inner/outer overcuts, vs intarsion operator result:
[image]

nfz, 2016-01-15:
Cool.  My test case works now.  Thanks.  I will play some more with this then update the wiki.  Its definitely a lot faster this way than the way I was doing it.  Greatly appreciate this.

narvf, 2016-01-18:
Great. I will check this in free time.
Thank you.

schef, 2016-01-22:
Hey guys. I'm just trying this out and am not sure how to use this when i have an mash object for example a cube 20x20x3 with inside hole which is 5x5x3. is there a way to apply overcuts to cam path?

vilemnovak, 2016-01-22:
Use curve objects.

schef, 2016-01-22:
Should i convert my whole object to curve? Or just use a 2d curve?

schef, 2016-01-26:
Are there some sugesstions to overcome this specific problem. I am trying to use blender only but am not sure how to make cornerovecut on the mesh objects. Any sugestions would be appreciated (also workaronds).

nfz, 2016-01-26:
Could you post an image of the mesh and the cam path?

schef, 2016-01-27:
So my tool is 3.5mm

[image]

vilemnovak, 2016-01-31:
this is a perfect case for doing it as a curve, not as a mesh object...
-Create a curve
-use overcut operator on it,
-use the result curve for your profile operation.
I am closing this as an "issue", since I consider this now more a support topic, which should be handled in the forum.

closed 2016-01-31 by vilemnovak