vilemduha/blendercam #21: Pocket operation?

narvf, 2016-01-15:
Is pocket operation working correctly? When i have a rectangle for example, and change pocket start position for example from outside to inside and export file, i have the same files. How works middle option? File seems to be the same.

What do you think about zigzag pocketing?  It looks much better on plexi for example, when ther are only straight lines and one around the shape  - on contour.

nfz, 2016-01-17:
Yep, I noticed the same thing but haven't looked at the code to see why.  I need to experiment with multiple curves that are inside each other, might have something to do with it.

nfz, 2016-01-17:
I had a look at the code.  Start positions outside/inside have not been implemented yet for pocket.  Only middle has been implemented.  Should probably disable "Start Position" option in the ui for now until the code gets implemented.

closed 2023-01-14 by pppalain