vilemduha/blendercam #22: NC library outdated

nfz, 2016-01-21:
I was investigating an issue with dwell gcode being generated by iso.py and have discovered it was fixed in Heekscnc many years ago (Sep 2013).  We should look at updating NC library since it looks like several other bugs were fixed in the last 3 years.

Some discussion of the issue is here:
http://blendercam.blogspot.fi/p/blog-page.html?place=msg%2Fblender-cam-forum%2F9xZe9V8S4iQ%2FgXKwWQjTAAAJ

nfz, 2016-01-21:
I forked the heekscnc nc on google code to here:
https://github.com/nfz/heekscnc/tree/master/nc

There is another NC on github but it is 3 years old.

vilemnovak, 2016-01-21:
The thing is, I also modified /fixed NC library. Main change is I added
support for rotary axes. We could commit these changes to heeks, then
update?
Dne 21. 1. 2016 8:13 napsal uživatel "Jeff Doyle" <notifications@github.com

nfz, 2016-01-21:
Ok, I could test it out in a branch and then do a pull request.

vilemnovak, 2016-01-22:
That would be great ;)

nfz, 2016-01-22:
I have create a NCupdate branch in my fork:

https://github.com/nfz/blendercam/tree/NCupdate

nfz, 2016-01-28:
Update is done.  So far everything seems to be working as before.  Dwell output now works correctly.  Going to spend another day testing all the post processors before I generate a pull request.

vilemnovak, 2016-02-01:
I am closing this, due to successfull merge ;)

closed 2016-02-01 by vilemnovak

nfz, 2016-02-01:
I was going to leave this open until more people got a chance to test it but if an issue does arise then a new issue can be opened.  The only post processor which has an issue is shopbot which never worked, even in Heekscnc.  I will get the shopbot working but there are other issues needing attention first ie documentation and fixing some more bugs.