vilemduha/blendercam #23: Can't find directory or file problem when trying to run simulation.

PaulSgrillo, 2016-01-21:
The following is all I get in the error report. The directory is there and the file is also. I can open the file in Krita and it looks fine. What have I done wrong? Didn't have th eproblem in previous version.
Also when trying to enable the addon received errors about not having msvcp140.dll and vcruntime140.dll on my system. Had to download them. Should they be included in the BlenderCam download?
progress{building paths from chunks}
3051
3051
3051
3051
0.2200002670288086
0.020000219345092773
progress{exporting gcode file}
10889
1
2.020002603530884
progress{total time 2.759%}
numpy to image _Operation_1_sim.exr
569 569

time 0.37000036239624023
Traceback (most recent call last):
  File "C:\Users\Paul\BlenderCAM\blender-2.76\2.76\scripts\addons\cam\ops.py", line 338, in execute
    utils.doSimulation(operation.name,[operation])
  File "C:\Users\Paul\BlenderCAM\blender-2.76\2.76\scripts\addons\cam\utils.py", line 870, in doSimulation
    i=bpy.data.images.load(iname)
RuntimeError: Error: Cannot read 'temp_cam\_Operation_1_sim.exr': No such file or directory

location: <unknown location>:-1

location: <unknown location>:-1

nfz, 2016-01-21:
Vilem is going to fix the problem with the missing dlls.  I ran into the same problem with the simulation crashing last night where the image file can't be found.  I haven't had a chance to investigate this problem yet but will look at it tonight.

Thanks for opening the issue.

PaulSgrillo, 2016-01-21:
Thanks, for quick reply. Felt like I was chasing my tail.

PaulSgrillo, 2016-01-21:
Didn't mean to close issue in case others have same issue and think they are doing something wrong.

nfz, 2016-01-22:
Yep, leave it open until the issue gets fixed and gets verified by you and others.

vilemnovak, 2016-01-22:
This is a strange issue - 
I blenderCAM the .blend file has to be saved somewhere to run simulation properly. But I guess this is not the case?
Maybe it can be something in the filepath, some special characters?

PaulSgrillo, 2016-01-22:
Don't know if it is related to the issue but why when opening the drop down in user prefs, the scene:Cam gcode generation reports as 0.8.0 for the revision number. Should this be 0.9.0?

nfz, 2016-01-23:
Yep, missed that. Now fixed in master.  Thanks.

closed 2023-01-14 by pppalain