vilemduha/blendercam #24: Profile-cutout error on generating path

narvf, 2016-01-24:
I've encountered an error when trying to generate cutout operation.
I've attached files.

[ornamentlog.txt](https://github.com/vilemnovak/blendercam/files/102471/ornamentlog.txt)

nfz, 2016-01-25:
The zip only has the svg but I will try it and see what happens.

nfz, 2016-01-25:
I was not able to import the ornament.svg into blender or into inkscape.

nfz, 2016-01-25:
Ok, found the problem with the zip.  Its a double un-zip.  The first extraction gives you ornament.svg and then I had to rename it to ornament.zip and then do a second extraction.  I now have ornament.blend and ornament.svg.

Very nice pattern.  This would make a beautiful v-carve which I will eventually do but I will first try cutout :).

nfz, 2016-01-25:
I tried the cutout operation without modifying your setup in the CAM panels and  I get the same error.  This type of error is usually a setup error either with the curves or with the CAM panel settings.  Some of your CAM panel settings raise some questions as to what your end goal is.

The first question is what are you trying to cut out in the design?  There is a curve border around the design and this is what is confusing Shapely (and me) causing it to abort.  
[image]

After removing the border and changing a few of the CAM panel settings, I was able to get a cutout but I am not sure this is what you want to do. You will end up with a lot of little pieces if the cutter goes full depth of the material.  Here is a pic of the resulting cutout path:
[image]

narvf, 2016-01-25:
Well, i would like to cut out all little pieces and i will have a square with a lot of holes - as a pattern for my sister.
Usually when i have a shape with holes i join them in Inkscape and then o create one operation to cut outside one shape and all shapes inside have added automatically cut out operaion to cut inside ;) and one big shape to cut outside.
Hard to exaplain in english :) Hope that helps.

nfz, 2016-01-25:
Ah, ok got it.  You are making a puzzle, cool.

nfz, 2016-01-25:
Something like this:
[image]

nfz, 2016-01-25:
If you separate the box curve into its own object then you can do what you intended.

narvf, 2016-01-25:
Well, not exactly :)
It will be something like this ;) I need to work on my English ;D
[image]

nfz, 2016-01-26:
No, after re-reading your post I realized I had misunderstood.  Now that I see what you want I would have selected cutout outside too.  I need to look at the code more to see why this failed.

nfz, 2016-01-26:
Using coutout online will work without crashing.  Cutout inside will crash.  It appears there is a problem in Blendercam with cutout when applied to your blend.

If I separate the square curve into another object and then use cutout outside on the square and cutout inside on the pattern then it works.  This is a work around until this gets fixed.

I made up a test case, a circle inside a circle curve, and it works fine.    I am rechecking your blend curves to see if there is a problem with the curves or there is a point when Blendercam/Shapely can't handle the complexity.

I have some projects coming up that will require cutting out complex shapes for stenciling so I have vested interest in getting this fixed now :).

nfz, 2016-01-26:
Did some research on the error messages Shapely was throwing and found the problem.  Some of the curves are malformed ie multiple control points overlaping.  It took a while to find them but there were 12 of them (I think).  This is what they look like when zoomed in:
[image]
I moved some of the control points since they were all on top each other.  Using the curve tool C-Remove Doubles sort of worked but it destroyed some of the curves so its no good.  The only way to fix was to manually search for them which was a pita.

narvf, 2016-01-26:
Ok, thanks. So there is no problem with blendercam but with svg file. Maybe i would be possible for blendercam to make a warning with human readable text like example "Please check your paths/curves" ?

nfz, 2016-01-26:
The svg file is fine, its the svg importer that creates the overlapping control points at sharp corners in the curve.  I loaded the svg in inkscape and its fine.  Run it through the blender svg importer script and the problems appear.

I will attach a zip of your corrected file latter in the day.

nfz, 2016-01-27:
I thought I had found the root problem for the multiple control points at the same position.  Each time I edit the curve more show up so it may not be the importer but something internal to blender.

narvf, 2016-01-27:
Ok then. We have a workaround to set cut-out on inside first and then oustide on rectangle.
Thanks for help.

closed 2016-02-22 by narvf