vilemduha/blendercam #27: Waterline strategy issues

filippsudanov, 2016-02-07:
[image]
[3020_cnc_end_switch_x_axis_plate.stl.zip](https://github.com/vilemnovak/blendercam/files/120865/3020_cnc_end_switch_x_axis_plate.stl.zip)

Here's a part, that has two pockets (that are open on the edges). In the right pocket the middle wall got processed correctly - it has waterlines. The other two walls miss the waterlines for some reasons. 
Also area filling in the right pocket did not work exactly right - the rightmost toolpath went up. The same toolpath in the left pocket is at the correct level. 
stl file is in the attached zip.

blendercam commit e16f2520369f13b6a197ec2276646f9f384849f2
blender version 2.76 i686
shapely 1.5.13
Ubuntu Linux 15.04 (Vivid Vervet) i686

P.S. In general I wonder, which strategy would be able to give best results for this part, so that walls of the pockets would be flat.

vilemnovak, 2016-02-12:
Hi, best strategy for this is pocketing - have 2d curves in several layers, use them for pocket operations, or maybe even drill operations. It's an overkill to use waterline strategy for a 2.5d object, and is not that precise as exact curves...
however, the bug report is valid, so I will keep it and check it when I have time. It's about the way the chunks get clamped when applying ambient.

FINThunderball, 2021-08-17:
Hi! 
We've been trying to configure blender - blenderCAM and Opencamlib -combination. 

For some reason blendercam says in CAM info & warnings that Opencamlib NOT installed. 

Why is this so? 
When tapping the "use experimental features on, it doesnt show the waterline options?

We also get error code: ....ops.py", line 496, in execute indexError: bpy_prop_collection[index]: index 0 out of range, size 0

Br
thunderball

FINThunderball, 2021-08-17:
We have used both Ubuntu in hyper v manager and Debian 11amd64 on oracle virtual box. 
br
Thunderball

closed 2023-01-14 by pppalain