vilemduha/blendercam #28: The libomp-dev package is missing in the distribution Debian Jessie

torvn77, 2016-02-07:
When you try to put to put the packages listed as dependencies OpenCAM I get a message saying that the package libomp-dev is missing in the repository.
Here is my /etc/apt/sources.list
`deb http://ftp.debian.org/debian/ jessie main  contrib non-free
deb-src http://ftp.debian.org/debian/ jessie main  contrib non-free
#deb http://mirror.yandex.ru/debian/ jessie main  contrib non-free
#deb-src http://mirror.yandex.ru/debian/ jessie main  contrib non-free

deb http://ftp.debian.org/debian/ jessie-backports main contrib non-free
deb-src http://ftp.debian.org/debian/ jessie-backports main contrib non-free
#deb http://mirror.yandex.ru/debian/ jessie-backports main contrib non-free
#deb-src http://mirror.yandex.ru/debian/ jessie-backports main contrib non-free

#deb http://ftp.debian.org/debian/ jessie-backports-sloppy main contrib non-free
#deb-src http://ftp.debian.org/debian/ jessie-backports-sloppy main contrib non-free

deb http://security.debian.org/ jessie/updates main contrib non-free
deb-src http://security.debian.org/ jessie/updates main contrib non-free

# jessie-updates, previously known as 'volatile'
deb http://ftp.debian.org/debian/ jessie-updates main contrib non-free
deb-src http://ftp.debian.org/debian/ jessie-updates main contrib non-free
#deb http://mirror.yandex.ru/debian/ jessie-updates main contrib non-free
#deb-src http://mirror.yandex.ru/debian/ jessie-updates main contrib non-free

deb http://ftp.debian.org/debian/ jessie-proposed-updates main contrib non-free
deb-src http://ftp.debian.org/debian/ jessie-proposed-updates main contrib non-free

###################################

#deb     http://ftp.debian.org/debian/ testing main  contrib non-free
#deb-src http://ftp.debian.org/debian/ testing main  contrib non-free

#deb     http://ftp.debian.org/debian/ testing-updates main  contrib non-free
#deb-src http://ftp.debian.org/debian/ testing-updates main  contrib non-free

#deb     http://ftp.debian.org/debian/ testing-proposed-updates main  contrib non-free
#deb-src http://ftp.debian.org/debian/ testing-proposed-updates main  contrib non-free

####################################
#deb     http://ftp.debian.org/debian/ unstable main  contrib non-free
#deb-src http://ftp.debian.org/debian/ unstable main  contrib non-free
`

ksuprynowicz, 2016-02-08:
It seems there's no libomp-dev in Debian Jessie. It's there in the Debian 'testing' version. 
Can you try using libiomp-dev? (it's available in 'stable' - Jessie repository):
    https://packages.debian.org/jessie/libdevel/libiomp-dev

Please let me know if it works

torvn77, 2016-02-08:
attempt to perform `make package` ended with the following error:  
'[ 97%] Building CXX object src/CMakeFiles/ocl.dir/ocl_algo.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
src/CMakeFiles/ocl.dir/build.make:100: recipe for target 'src/CMakeFiles/ocl.dir/ocl_algo.o' failed
make[2]: *** [src/CMakeFiles/ocl.dir/ocl_algo.o] Error 4
CMakeFiles/Makefile2:131: recipe for target 'src/CMakeFiles/ocl.dir/all' failed
make[1]: *** [src/CMakeFiles/ocl.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
'

PaulSgrillo, 2016-02-11:
After several efforts of removing and reinstalling the OpenCamLib, I keep getting the same errors. Looks like I need some assistance in setting things up correctly. If you get a chance and can help me out, I would be greatfull.
Paul

Waterline Attempt to generate tool path on simple 1/2"H by 4"D disk

Traceback (most recent call last):
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/opencamlib/oclWaterline.py", line 3, in <module>
    import camvtk
  File "/usr/local/lib/python2.7/dist-packages/camvtk.py", line 9, in <module>
    import vtk
  File "/usr/lib/python2.7/dist-packages/vtk/__init__.py", line 39, in <module>
    from vtkCommonCore import *
  File "/usr/lib/python2.7/dist-packages/vtk/vtkCommonCore.py", line 1, in <module>
    from vtkCommonCorePython import *
ImportError: No module named vtkCommonCorePython
Traceback (most recent call last):
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/ops.py", line 214, in execute
    utils.getPath(context,o)
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/utils.py", line 3233, in getPath
    getPath3axis(context,operation)
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/utils.py", line 2614, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 170, in oclGetWaterline
    waterlineChunksFromCSV( operation, chunks )
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 145, in waterlineChunksFromCSV
    csv_file = open(tempfile.gettempdir() + '/oclWaterline' + str(wl_index) + '.txt', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/oclWaterline0.txt'

location: <unknown location>:-1

location: <unknown location>:-1

Parallel Attempt to generate tool path on simple 1/2"H by 4"D disk
update strategy
update exact 
update exact 
progress{building path pattern}
progress{0.8509140014648438}
Traceback (most recent call last):
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/opencamlib/oclSample.py", line 4, in <module>
    import camvtk
  File "/usr/local/lib/python2.7/dist-packages/camvtk.py", line 9, in <module>
    import vtk
  File "/usr/lib/python2.7/dist-packages/vtk/__init__.py", line 39, in <module>
    from vtkCommonCore import *
  File "/usr/lib/python2.7/dist-packages/vtk/vtkCommonCore.py", line 1, in <module>
    from vtkCommonCorePython import *
ImportError: No module named vtkCommonCorePython
Traceback (most recent call last):
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/ops.py", line 214, in execute
    utils.getPath(context,o)
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/utils.py", line 3233, in getPath
    getPath3axis(context,operation)
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/utils.py", line 2591, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/utils.py", line 327, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 111, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/paul/Documents/Blender276Cam/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 48, in chunkPointSamplesFromCSV
    csv_file = open(tempfile.gettempdir()+'/ocl_chunk_samples.txt','r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: <unknown location>:-1

location: <unknown location>:-1

ksuprynowicz, 2016-02-11:
Hi torvn77,

It looks like problem with GCC-4.9, and not OpenCAMLib. Try changing GCC version before building:

    make clean
    cmake-gui .

then in cmake-gui change compiler to another version (there's gcc-4.8/g++-4.8 and possibly others available in official package repository, install it with aptitude)

In cmake-gui you have to change CMAKE_CXX COMPILER and CMAKE_C_COMPILER settings to /usr/bin/g++-4.8 and /usr/bin/gcc-4.8 respectively.
Then click 'Configure' and 'Generate' button.

What architecture do you use? Can you type result of:

    uname -r

ksuprynowicz, 2016-02-11:
Hi Paul,

Your problem seems to be the same as in this bug:
https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1354127

Which version of python-vtk do you have(you can check it in aptitude)? Also only VTK6 should be installed (VTK5 may cause problems).
Did you install any python packages through pypi package manager?

PaulSgrillo, 2016-02-11:
Hi ksuprynowicz,
Thanks for getting back.
I have python-vtk6 installed. vtk5 was there and I removed it still the same issue. Installed python thru Synaptic Package Manager. There seem to be several 2.7 versions installed 2.75.5 default ver and 2.7.6 . Is this a possible. 
One thing really confusing me in the error messages are: ImportError: No module named vtkCommonCorePython, looke for this and it isn't there. Also; FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt' , where would one look for a tmp directory.

ksuprynowicz, 2016-02-12:
The '/tmp/ocl_chunk_samples.txt' file gets created by python2.7 script using OpenCAMLib. It should show up when script executes correctly, and I think it will, when the problem with vtkCommonCorePython is solved. It works in Debian testing, so should be possible in Debian Jessie too.

Please check if following files in /usr/lib/python2.7/dist-packages/vtk/ exist(and if they are symlinks, do they point to valid file?):

    vtkCommonCore.py
    vtkCommonCore.pyc
    vtkCommonCorePython.x86_64-linux-gnu.so

If they are there, open new terminal, and try adding /usr/lib/python2.7/dist-packages/vtk/ to the Python path:

    export PYTHONPATH="$PYTHONPATH:/usr/lib/python2.7/site-packages/vtk/"

then form the same terminal execute blender( Python path will be chandged only for this terminal).

Do you have a local python2.7 instalation? It could be used instead of the one installed by synaptic package manager and then fail to find python-vtk6.

If everything fails, maybe you can try installing Debian Testing as dual boot or upgrading from Jessie to Testing? This way you will have recent version of all the packages, including blender available from synaptic. Stable versions of Debian can get quite outdated, and Testing is definitely stable enough for production use.

torvn77, 2016-02-12:
Hi ksuprynowicz,
 uname -r
4.2.0-0.bpo.1-amd64
 c++ -v  && cc -v
gcc version 4.9.2 (Debian 4.9.2-10)

Downgrade GCC or remain the same?

ksuprynowicz, 2016-02-12:
There's no need to downgrade. You can have both 4.8 and 4.9 installed and select 4.8 when using cmake-gui as I described in earlier post. Let me know how this goes.

ksuprynowicz, 2016-02-13:
I've updated install instructions. Now they'll work for fresh Debian install, there's no need to download anything. I've tested it on VitrualBox virtual machine, and everything compiles and works perfectly( I've tested both drop cutter sampling and waterlines ).
Maybe you can set up a virtual machine and give it a try? (Remember to turn on 3d acceleration in VirtualBox settings)
The whole virtual machine, BlenderCAM and OpenCAMLib setup takes about an hour.
Here are install instructions:

https://github.com/vilemnovak/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib

I've used this image:

http://cdimage.debian.org/debian-cd/8.3.0/amd64/iso-cd/debian-8.3.0-amd64-netinst.iso

torvn77, 2016-02-14:
Hi ksuprynowicz 
Under real in Debian Jessie (as I understand it, you call it an 8.3) installation attempt ended with this:  
`user@ggg:~/Git/opencamlib/builds$ make  
[ 46%] Built target libocl  
[ 49%] Built target ocl_common  
[ 60%] Built target ocl_cutters  
[ 74%] Built target ocl_geo  
[ 87%] Built target ocl_algo  
[ 92%] Built target ocl_dropcutter  
[ 94%] Building CXX object src/CMakeFiles/ocl.dir/ocl_cutters.o  
c++: internal compiler error: Killed (program cc1plus)  
Please submit a full bug report,  
with preprocessed source if appropriate.  
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.  
src/CMakeFiles/ocl.dir/build.make:54: recipe for target 'src/CMakeFiles/ocl.dir/ocl_cutters.o' failed  
make[2]: *** [src/CMakeFiles/ocl.dir/ocl_cutters.o] Error 4  
CMakeFiles/Makefile2:131: recipe for target 'src/CMakeFiles/ocl.dir/all' failed  
make[1]: *** [src/CMakeFiles/ocl.dir/all] Error 2  
Makefile:137: recipe for target 'all' failed  
make: *** [all] Error 2  
user@ggg:~/Git/opencamlib/builds$ `

ksuprynowicz, 2016-02-15:
I've tested with exactly the same release (8.3), 64bit, fresh install, gcc-4.9 and both compilation and BlenderCAM integration works.(I've tested on 2xOpteron 6174, 32GB ECC mem.)

Maybe this is hardware issue? Compilation stopping, probably with segfault on two different files pionts to that. The three most probable possibilities are:

- Your processor needs microcode update to be applied at system start:

https://wiki.debian.org/Microcode

- Not enough swap memory/swap is not enabled, you can check with:

    swapon -s

- Problems with memory. You can check by installing memtest86+ (not memtest86) package, rebooting computer and selecting it in boot menu. Then you have to run it for a few hours and see if the errors show up(it has to pass all kinds of tests at least once)

ksuprynowicz, 2016-02-15:
Other machines I've tested are 2x Intel Xeon 5670, 32GB ECC mem. and AMD Phenom II 945 with 16GB ECC mem.

torvn77, 2016-02-17:
Hi ksuprynowicz
Put the firmware and ran htop to monitor RAM.
And really, at the end of compilation is almost more than 1 Giga ate, so if I opened the browser it would crash.
But it is strange that process immediately collapsing, without the usual out-of-memory performance.

In General, compilation and installation finished successfully.
put in a blender box, a spiral on the sphere is generated.
Well, sort of.

ksuprynowicz, 2016-02-17:
If you have enough swap, processes should never crash from lack of memory on 64 bit architecture. I've seen such problem a few times, and it was always fault of memory module.

Please test the waterline and parallel sampling with 'Use OpenCAMLib' for waterline, and both 'Use OpenCAMLib' and 'Use Exact Mode' for parallel, and let me know if it works.

torvn77, 2016-02-22:
I don't really understand what you want, here's a screenshot c `EXPEREMENTAL Waterline`
OpenCAMLib is enabled.
[image]

ksuprynowicz, 2016-02-22:
Exactly like this, but in your screenshot OpenCAMLib is disabled :)

>I don't really understand what you want, here's a screenshot c 
`EXPEREMENTAL Waterline`

torvn77, 2016-02-22:
чекбокс
I noticed my mistake :)
Here is a screenshot with the "waterline" and activated opencamlib
but here's the checkbox exact mode I have not found, for it is this trajectory?
[image]

torvn77, 2016-02-22:
Found the checkbox for exact mode
[image]

torvn77, 2016-02-22:
I noticed my mistake.
The checkbox exact regime for WATER LINES not found

[image]

ksuprynowicz, 2016-02-23:
The 'exact mode' checkbox is not needed for waterlines. From the screenshots you've sent everything works ok, so I'm closing the issue.

closed 2016-02-23 by ksuprynowicz

torvn77, 2016-02-23:
Thank you very much, goodbye