vilemduha/blendercam #31: autogenerate bridges generates an exception

nfz, 2016-02-11:
Tried to auto generate bridges when doing a profile cutout of a simple circle curve and an exception occurs in Shapely.  I haven't investigated any further but someone else also ran into the problem so figure its worth looking into now.

https://groups.google.com/forum/#!category-topic/blender-cam-forum/8Ec7X12qNn8

The traceback that I get:
```
Traceback (most recent call last):
  File "/home/nfz/SoftwareDevelopment/blendercam1/scripts/addons/cam/ops.py", line 690, in execute
    utils.addAutoBridges(o)
  File "/home/nfz/SoftwareDevelopment/blendercam1/scripts/addons/cam/utils.py", line 1961, in addAutoBridges
    d1 = c.project(sgeometry.Point(maxx+1000, (maxy+miny)/2.0))
  File "/home/nfz/SoftwareDevelopment/blender-git/build/bin/2.76/python/lib/python3.5/site-packages/shapely/impl.py", line 30, in wrapper
    return func(*args, **kwargs)
  File "/home/nfz/SoftwareDevelopment/blender-git/build/bin/2.76/python/lib/python3.5/site-packages/shapely/geometry/base.py", line 665, in project
    return op(self, other)
  File "/home/nfz/SoftwareDevelopment/blender-git/build/bin/2.76/python/lib/python3.5/site-packages/shapely/linref.py", line 17, in __call__
    self._validate_line(this)
  File "/home/nfz/SoftwareDevelopment/blender-git/build/bin/2.76/python/lib/python3.5/site-packages/shapely/linref.py", line 13, in _validate_line
    raise TypeError("Only linear types support this operation")
TypeError: Only linear types support this operation
```

vilemnovak, 2016-04-11:
I need a .blend file to check the bug, for me it works here - you have to have an operation with a curve object/objects.

nfz, 2016-05-28:
Found the problem.  I was using Shapely 1.4 which does not allow LinearRings to do a project but in 1.5 it does.

nfz, 2016-05-28:
All is good.

closed 2016-05-28 by nfz