vilemduha/blendercam #32: opencamlib not working

kylotron, 2016-02-12:
Hi...

i have compiled opencamlib... but not working on blendercam!

Help me! please!

> bpy.ops.object.select_all(action='DESELECT')
> bpy.ops.object.duplicate(linked=False)
> bpy.ops.object.transform_apply(location=True, rotation=True, scale=True)
> bpy.ops.transform.resize(value=(1000, 1000, 1000), constraint_axis=(False, False, False), constraint_orientation='GLOBAL', mirror=False, proportional='DISABLED', proportional_edit_falloff='SMOOTH', proportional_size=1, snap=False, snap_target='CLOSEST', snap_point=(0, 0, 0), snap_align=False, snap_normal=(0, 0, 0), texture_space=False, release_confirm=False)
> bpy.ops.object.transform_apply(location=True, rotation=True, scale=True)
> bpy.ops.object.delete()
> Traceback (most recent call last):
>   File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/ops.py", line 218, in execute
>     utils.getPath(context,o)
>   File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 3236, in getPath
>     getPath3axis(context,operation)
>   File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 2614, in getPath3axis
>     oclGetWaterline(o, chunks)
>   File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 166, in oclGetWaterline
>     if os.path.isdir(bpy.utils.script_path_pref()+"/addons/cam/opencamlib"):
> TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
> 
> location: <unknown location>:-1

ksuprynowicz, 2016-02-12:
Hi!
To fix this select 'User Preferences' form 'File' menu, then 'File' tab in preferences. There you have to enter BlenderCAM scripts directory into 'Scripts' field. In you case it'll be:

     /opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts

Please let me know if it works - I'll add it to wiki then.

kylotron, 2016-02-12:
This is the new error:

> Traceback (most recent call last):
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 3236, in getPath
    getPath3axis(context,operation)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 2614, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 170, in oclGetWaterline
    waterlineChunksFromCSV( operation, chunks )
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 145, in waterlineChunksFromCSV
    csv_file = open(tempfile.gettempdir() + '/oclWaterline' + str(wl_index) + '.txt', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/oclWaterline0.txt'

location: <unknown location>:-1

kylotron, 2016-02-12:
new .blend object and error:

> Traceback (most recent call last):
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 3236, in getPath
    getPath3axis(context,operation)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 2614, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 170, in oclGetWaterline
    waterlineChunksFromCSV( operation, chunks )
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 145, in waterlineChunksFromCSV
    csv_file = open(tempfile.gettempdir() + '/oclWaterline' + str(wl_index) + '.txt', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/oclWaterline0.txt'

location: <unknown location>:-1

[Test.blend.zip](https://github.com/vilemnovak/blendercam/files/128682/Test.blend.zip)

ksuprynowicz, 2016-02-12:
I've tested the file you included with Blender CAM 0.9.0 release and blender 2.76b, and it works (waterlines are computed correctly, only there's no fill yet - has to be added).

Can you run blender from terminal and post the output that shows up?
When you compute operation, it should be like this:

```
read blend: /tmp/2/Test.blend
0.0

BPC::setSTL() Building kd-tree... bucketSize=1..BPC::setSTL() root->build()...done.
BPC::setSTL() Building kd-tree... bucketSize=1..BPC::setSTL() root->build()...done.
 Waterline::init_fibers()
BatchPushCutter3 with 1020 fibers and 140 triangles.
 cutter = CylCutter (d=3, L=70)

0%   10   20   30   40   50   60   70   80   90   100%
|----|----|----|----|----|----|----|----|----|----|
Number of OpenMP threads = 24
*******************************************************
BatchPushCutter3 done.
BatchPushCutter3 with 1020 fibers and 140 triangles.
 cutter = CylCutter (d=3, L=70)

0%   10   20   30   40   50   60   70   80   90   100%
|----|----|----|----|----|----|----|----|----|----|
Number of OpenMP threads = 24
*********************************************************
BatchPushCutter3 done.
Weave...
 SimpleWeave::build()... 
done.
Weave::face traverse()... traversing graph with 6000 cl-points
done.
Weave::get_loops()...done.
-9.99999977648
```

It continues like this for some time, and then upon completion:

```
~Waterline_py()
~Waterline(): subOp.size()= 2
progress{building paths from chunks}
4040
1960
4040
1960
4040
1960
4040
1960
4040
4040
4040
0.14377546310424805
0.0012640953063964844
progress{total time 1.221%}
```

kylotron, 2016-02-12:
this is the error from terminal:

> CAD /opt/blender-2.76b-linux-glibc211-x86_64 $ ./blender
connect failed: No such file or directory
Read new prefs: /home/kylotron/.config/blender/2.76/config/userpref.blend
found bundled python: /opt/blender-2.76b-linux-glibc211-x86_64/2.76/python
read blend: /home/kylotron/Scrivania/Modelli/Test.blend
	WM_OT_context_toggle
	WM_OT_context_toggle_enum
	WM_OT_context_cycle_int
	WM_OT_context_cycle_enum
	WM_OT_context_cycle_array
	WM_OT_context_menu_enum
Traceback (most recent call last):
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/oclWaterline.py", line 3, in <module>
    import camvtk
  File "/usr/local/lib/python2.7/dist-packages/camvtk.py", line 9, in <module>
    import vtk
  File "/usr/lib/python2.7/dist-packages/vtk/__init__.py", line 39, in <module>
    from vtkCommonCore import *
  File "/usr/lib/python2.7/dist-packages/vtk/vtkCommonCore.py", line 1, in <module>
    from vtkCommonCorePython import *
ImportError: No module named vtkCommonCorePython
Traceback (most recent call last):
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 3236, in getPath
    getPath3axis(context,operation)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/utils.py", line 2614, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 170, in oclGetWaterline
    waterlineChunksFromCSV( operation, chunks )
  File "/opt/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 145, in waterlineChunksFromCSV
    csv_file = open(tempfile.gettempdir() + '/oclWaterline' + str(wl_index) + '.txt', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/oclWaterline0.txt'

location: <unknown location>:-1

location: <unknown location>:-1

vtk is installed :(

kylotron, 2016-02-12:
is possible, in wiki, add all steps (step by step) for installing all libraries for blendercam?

Numpy, Shapely, Polygon, OpenCamlib, Vtk, etc... etc...  etc...

kylotron, 2016-02-13:
I have cleaned my PC and have reinstalled blender, blendercam, numpy, polygon, shapely and opencamlib.
This is the error:

`Traceback (most recent call last):
  File "/opt/Blender2.77/2.76/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/opt/Blender2.77/2.76/scripts/addons/cam/utils.py", line 3215, in getPath
    chd=getChangeData(operation)
  File "/opt/Blender2.77/2.76/scripts/addons/cam/utils.py", line 194, in getChangeData
    obs=[bpy.data.objects[o.object_name]]
KeyError: 'bpy_prop_collection[key]: key "" not found'`

and this error is for ALL strategies and ALL .blend files!

Other step... i have cleaned the configuration file of blender and temp files of linux....

And all strategies function correctly but  i tested NOT witch opencamlib!

```
update opencamlib 
changing sampling resolution to 0.000122
Traceback (most recent call last):
  File "/opt/Blender2.77/2.76/scripts/addons/cam/opencamlib/oclWaterline.py", line 3, in <module>
    import camvtk
  File "/usr/local/lib/python2.7/dist-packages/camvtk.py", line 9, in <module>
    import vtk
ImportError: No module named vtk
Traceback (most recent call last):
  File "/opt/Blender2.77/2.76/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/opt/Blender2.77/2.76/scripts/addons/cam/utils.py", line 3236, in getPath
    getPath3axis(context,operation)
  File "/opt/Blender2.77/2.76/scripts/addons/cam/utils.py", line 2614, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/opt/Blender2.77/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 170, in oclGetWaterline
    waterlineChunksFromCSV( operation, chunks )
  File "/opt/Blender2.77/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 145, in waterlineChunksFromCSV
    csv_file = open(tempfile.gettempdir() + '/oclWaterline' + str(wl_index) + '.txt', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/oclWaterline0.txt'
location: <unknown location>:-1
location: <unknown location>:-1
``

kylotron, 2016-02-13:
`FileNotFoundError: [Errno 2] No such file or directory: '/tmp/oclWaterline0.txt'`

???

ksuprynowicz, 2016-02-13:
I've updated install instructions. Now they'll work for fresh Debian install, there's no need to download anything. I've tested it on VitrualBox virtual machine, and everything compiles and works perfectly( I've tested both drop cutter sampling and waterlines ).
Maybe you can set up a virtual machine and give it a try? (Remember to turn on 3d acceleration in VirtualBox settings)
The whole virtual machine, BlenderCAM and OpenCAMLib setup takes about an hour.
Here are install instructions:

https://github.com/vilemnovak/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib

I've used this image:

http://cdimage.debian.org/debian-cd/8.3.0/amd64/iso-cd/debian-8.3.0-amd64-netinst.iso

kylotron, 2016-02-13:
libomp-dev or libompl-dev ?

libomp-dev is not present!

ksuprynowicz, 2016-02-13:
In Debian Stable it's libiomp-dev, in Testing it's libomp-dev. Libompl-dev is something entirely different.

kylotron, 2016-02-13:
ok ;)

kylotron, 2016-02-16:
Yes... Debian 8.3 + normal installation of blender+blendercam etc (by wiki), function perfectly!

kylotron, 2016-02-16:
A new installation (by wiki) on Mint 17.3 (blender 2.69)  with libiomp-dev (not libomp-dev) return this error:

`Traceback (most recent call last):
  File "/usr/share/blender/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/usr/share/blender/scripts/addons/cam/utils.py", line 3215, in getPath
    chd=getChangeData(operation)
  File "/usr/share/blender/scripts/addons/cam/utils.py", line 194, in getChangeData
    obs=[bpy.data.objects[o.object_name]]
KeyError: 'bpy_prop_collection[key]: key "" not found'

location: <unknown location>:-1

`

ksuprynowicz, 2016-02-17:
I tried both libomp-dev and libiomp-dev, and it's ok. I think that Blender 2.69 may not be compatible with most recent versions of BlenderCAM due to API changes between 2.69 and 2.7x. Can you try Blender 2.74 or newer on Mint?

kylotron, 2016-02-17:
Linux Mint 17.3 Cinnamon + Blender 2.76b + BlenderCam + all libraries... 

`Traceback (most recent call last):
  File "/usr/share/blender/2.76/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/usr/share/blender/2.76/scripts/addons/cam/utils.py", line 3236, in getPath
    getPath3axis(context,operation)
  File "/usr/share/blender/2.76/scripts/addons/cam/utils.py", line 2614, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/usr/share/blender/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 166, in oclGetWaterline
    if os.path.isdir(bpy.utils.script_path_pref()+"/addons/cam/opencamlib"):
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'

location: <unknown location>:-1

`

ksuprynowicz, 2016-02-17:
Ok, this is better :)

Probably the only thing you have to do now, it setting 'scripts' path in blender preferences, as described in 'Using BlenderCAM with OpenCAMLib' Wiki:

> Then you have to run Blender from your distribution and set 'Scripts' directory in 'User preferences/File' to /home/yourusername/blendercam/scripts/

or other path depending on where you have BlenderCAM installed.

Let me know if it works.

kylotron, 2016-02-17:
No :( Not Working!

ksuprynowicz, 2016-02-17:
Is the error message still the same?:
    if os.path.isdir(bpy.utils.script_path_pref()+"/addons/cam/opencamlib"):
    TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'

After restarting blender(restart is necessary) settings should be like this:

[image]

kylotron, 2016-02-17:
[image]

I have this error in Linux Mint 17.3 Cinnamon!

kylotron, 2016-02-17:
I have removed all blendercam script in /usr/share/blender/2.76/script and i have set a new path
But... the error is this!

```
Traceback (most recent call last):
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 3236, in getPath
    getPath3axis(context,operation)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 2614, in getPath3axis
    oclGetWaterline(o, chunks)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 170, in oclGetWaterline
    waterlineChunksFromCSV( operation, chunks )
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 145, in waterlineChunksFromCSV
    csv_file = open(tempfile.gettempdir() + '/oclWaterline' + str(wl_index) + '.txt', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/oclWaterline0.txt'

location: <unknown location>:-1
```

ksuprynowicz, 2016-02-17:
Can you start the blender form terminal and post terminal output? Blender only shows errors on Python 3.3 side, OpenCAMLib output is only visible in terminal.

kylotron, 2016-02-17:
[image]

ksuprynowicz, 2016-02-17:
Seems like problem with VTK or vtk-python installation.
To verify if VTK works, open terminal, and run python2.7 console:

    python2.7

Then in console:

    import vtk

And post any errors if they show up

kylotron, 2016-02-17:
[image]

yes... vtk are a problem!

kylotron, 2016-02-17:
now, i have installed python-vtk 

`sudo apt-get install python-vtk`

and...
[image]

but blendercam not work!

[image]

ksuprynowicz, 2016-02-17:
Did you install VTK and vtk-python via APT (like aptitude or Synaptic package manager)?

kylotron, 2016-02-17:
installed vtk from aptitude.
but the error is same!

[image]

ksuprynowicz, 2016-02-17:
Can you check the version of vtk and python-vtk?
In terminal(as a regular user, not root) run following command:

    aptitude versions vtk

kylotron, 2016-02-17:
```
kylotron@kylotro-CAD ~ $ aptitude versions vtk
Pachetto libvtk-java:                   
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk-java:i386:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5-dev:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5-dev:i386:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5-qt4-dev:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5-qt4-dev:i386:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5.8:
i A 5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5.8-qt4:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5.8-qt4:i386:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk5.8:i386:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto libvtk6:
p   6.0.0-6                                       trusty                    500 

Pachetto libvtk6-dev:
p   6.0.0-6                                       trusty                    500 

Pachetto libvtk6-dev:i386:
p   6.0.0-6                                       trusty                    500 

Pachetto libvtk6-java:
p   6.0.0-6                                       trusty                    500 

Pachetto libvtk6-java:i386:
p   6.0.0-6                                       trusty                    500 

Pachetto libvtk6:i386:
p   6.0.0-6                                       trusty                    500 

Pachetto libvtkedge:
p   0.2.0~20110819-2                              trusty                    500 

Pachetto libvtkedge-dev:
p   0.2.0~20110819-2                              trusty                    500 

Pachetto libvtkedge-dev:i386:
p   0.2.0~20110819-2                              trusty                    500 

Pachetto libvtkedge:i386:
p   0.2.0~20110819-2                              trusty                    500 

Pachetto libvtkgdcm-cil:
p   2.2.4-1.1ubuntu4                              trusty                    500 

Pachetto libvtkgdcm-cil:i386:
p   2.2.4-1.1ubuntu4                              trusty                    500 

Pachetto libvtkgdcm-java:
p   2.2.4-1.1ubuntu4                              trusty                    500 
[... 78 lines cut ...]

Pachetto tcl-vtk:i386:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto vtk-doc:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto vtk-examples:
p   5.8.0-14.1ubuntu3                             trusty                    500 

Pachetto vtk6:
p   6.0.0-6                                       trusty                    500 

Pachetto vtk6-doc:
p   6.0.0-6                                       trusty                    500 

Pachetto vtk6-examples:
p   6.0.0-6                                       trusty                    500 

Pachetto vtk6:i386:
p   6.0.0-6                                       trusty                    500 

Pachetto vtkdata:
p   5.8.0-1                                       trusty                    500 
```

ksuprynowicz, 2016-02-17:
The version of OpenCAMLib in my github repository is patched for VTK6 and wont work with VTK5.
Your version of Linux Mint imports VTK5 by default.
The easiest way to make it work is to get older version of OpenCAMLib from original repository:

    https://github.com/aewallin/opencamlib.git

kylotron, 2016-02-17:
yeessssssssssssssssssss!

[image]

ksuprynowicz, 2016-02-18:
It's nice to see that it works :)
I'll add information about VTK5 to the Wiki

closed 2016-02-18 by ksuprynowicz

Filippide77, 2016-11-05:
Hi I have a problem with opencamlib.... My system is Debian 3.16.36-1+deb8u1, Blendercam 2.76b. I think to have installed VTK6 and I'm trying to install OpenCAMLib from https://github.com/ksuprynowicz/opencamlib.git.  
davide@debian:~$ aptitude versions vtk
Package libvtk-dicom-java:          
p   0.5.5-2                                       stable                    500 

Package libvtk-dicom0.5:
p   0.5.5-2                                       stable                    500 

Package libvtk-dicom0.5-dev:
p   0.5.5-2                                       stable                    500 

Package libvtk-java:
p   5.8.0-17.5                                    stable                    500 

Package libvtk5-dev:
p   5.8.0-17.5                                    stable                    500 

Package libvtk5-qt4-dev:
p   5.8.0-17.5                                    stable                    500 

Package libvtk5.8:
idA 5.8.0-17.5                                    stable                    500 

Package libvtk5.8-qt4:
p   5.8.0-17.5                                    stable                    500 

Package libvtk6-dev:
i   6.1.0+dfsg2-6                                 stable                    500 

Package libvtk6.1:
i A 6.1.0+dfsg2-6                                 stable                    500 

Package libvtkgdcm-cil:
p   2.4.4-3+deb8u1                                stable                    500 

Package libvtkgdcm-java:
p   2.4.4-3+deb8u1                                stable                    500 

Package libvtkgdcm-tools:
p   2.4.4-3+deb8u1                                stable                    500 

Package libvtkgdcm2-dev:
p   2.4.4-3+deb8u1                                stable                    500 

Package libvtkgdcm2.4:
p   2.4.4-3+deb8u1                                stable                    500 

Package lvtk-dev:
p   1.2.0~dfsg0-1                                 stable                    500 

Package lvtk-doc:
p   1.2.0~dfsg0-1                                 stable                    500 

Package lvtk-examples:
p   1.2.0~dfsg0-1                                 stable                    500 

Package lvtk-tools:
p   1.2.0~dfsg0-1                                 stable                    500 

Package php5-vtkgdcm:
p   2.4.4-3+deb8u1                                stable                    500 

Package python-pyvtk:
p   0.4.74-3                                      stable                    500 

Package python-vtk:
p   5.8.0-17.5                                    stable                    500 

Package python-vtk-dicom:
p   0.5.5-2                                       stable                    500 
[... 100 lines cut ...]
update exact 
update exact 
progress{building path pattern}
progress{0.015167236328125}
python2.7: can't open file '/home/davide/opencamlib/scripts/addons/cam/opencamlib/oclSample.py': [Errno 2] No such file or directory
Traceback (most recent call last):
  File "/home/davide/CNC/BlenderCam/2.76/scripts/addons/cam/ops.py", line 214, in execute
    utils.getPath(context,o)
  File "/home/davide/CNC/BlenderCam/2.76/scripts/addons/cam/utils.py", line 3233, in getPath
    getPath3axis(context,operation)
  File "/home/davide/CNC/BlenderCam/2.76/scripts/addons/cam/utils.py", line 2591, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/davide/CNC/BlenderCam/2.76/scripts/addons/cam/utils.py", line 327, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/davide/CNC/BlenderCam/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 111, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/davide/CNC/BlenderCam/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 48, in chunkPointSamplesFromCSV
    csv_file = open(tempfile.gettempdir()+'/ocl_chunk_samples.txt','r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: <unknown location>:-1

location: <unknown location>:-1

What can I do?

Filippide77, 2016-11-06:
Ok I have found the solution I must chose the folder script:
/home/davide/CNC/BlenderCam/2.76/scripts/
I had understood to must select
 /home/davide/opencamlib/scripts/
but in this case it not work

ksuprynowicz, 2016-11-07:
Hi Filippide77!
Does it work correctly with proper script directory?

Filippide77, 2016-11-07:
Yes, it seems to function properly....