vilemduha/blendercam #33: simpliying and speeding gcode?

ecl66, 2016-02-28:
Hello to the great development Team behind Blendercam.

I was wondering if there would be a way to speed and simplify the gcode generated. Here is an example using block-all:

N2100 G01 Y22.5
N2110 G01 Y27.5
N2120 G01 Y32.5
N2130 G01 Y32.999
N2140 G01 X22
N2150 G01 X27
N2160 G01 X32
N2170 G01 X32.999
N2180 G01 Y28
N2190 G01 Y23
N2200 G01 Y18
N2210 G01 Y17
N2220 G01 X28
N2230 G01 X24.333

It could best be generated as:

N2100 G01 Y22.5
N2130 G01 Y32.999
N2140 G01 X22
N2170 G01 X32.999
N2180 G01 Y28
N2210 G01 Y17
N2220 G01 X28
N2230 G01 X24.333

Am I missing something in the settings perhaps?

Many thanks and please keep on carrying such a good work (btw, sorting is working well now, still very slow on complex models but that's an issue with python I guess...).

Cheers

vilemnovak, 2016-02-28:
Hello, 
can you provide a file? This kind of  optimization should be allready present.

ecl66, 2016-02-29:
N10 (GCode created using the HeeksCNC Mach3 post processor)
N20 (/home/ecl/fine.ngc)
N30 G21
N40 (G-code generated with BlenderCAM and NC library)
N50 G17 G90
N60 (Tool change - D = 2.0 mm  type BALL flutes 2)
N70 G43H1
N80 T1M06
N90 G04 P5
N100 G00 X0 Y0 Z5 S12000 M03
N110 G00 X0.1 Y0.299
N120 G01 Z-0.5 F1200
N130 G01 X6.5 Y0.299 F2000
N140 G01 X9.7 Y0.299
N150 G01 X12.9 Y0.299
N160 G01 X18.1 Y0.299
N170 G01 X19.3 Y0.299
N180 G01 X20.5 Y0.3
N190 G01 X21.7 Y0.3
N200 G01 X22.9 Y0.3
N210 G01 X24.1 Y0.3
N220 G01 X24.9 Y0.3
N230 G01 Y0.7
N240 G01 X23.7
N250 G01 X22.5
N260 G01 X21.3
N270 G01 X20.1
N280 G01 X18.9
N290 G01 X17.7
N300 G01 X16.5
N310 G01 X15.3
N320 G01 X12.1
N330 G01 X8.9
N340 G01 X5.7
N350 G01 X2.5
N360 G01 X0.1
N370 G01 Y1.1
N380 G01 X6.5
N390 G01 X9.7
N400 G01 X12.9
N410 G01 X18.1
N420 G01 X19.3
N430 G01 X20.5
N440 G01 X21.7
N450 G01 X22.9
N460 G01 X24.1
N470 G01 X24.9
N480 G01 Y1.5
N490 G01 X23.7
N500 G01 X22.5
N510 G01 X21.3
N520 G01 X20.1
N530 G01 X18.9
N540 G01 X17.7
N550 G01 X16.5
N560 G01 X15.3
N570 G01 X12.1
N580 G01 X8.9
N590 G01 X5.7
N600 G01 X2.5
N610 G01 X0.1
N620 G01 Y1.9
N630 G01 X6.5
N640 G01 X9.7
N650 G01 X12.9
N660 G01 X18.1
N670 G01 X19.3
N680 G01 X20.5
N690 G01 X21.7
N700 G01 X22.9
[... 4383 lines cut ...]
N44540 G01 X18.9
N44550 G01 X17.7
N44560 G01 X16.5
N44570 G01 X15.3
N44580 G01 X12.1
N44590 G01 X8.9
N44600 G01 X5.7
N44610 G01 X2.5
N44620 G01 X0.1
N44630 G01 Y14.7
N44640 G01 X6.5
N44650 G01 X9.7
N44660 G01 X12.9
N44670 G01 X18.1
N44680 G01 X19.3
N44690 G01 X20.5
N44700 G01 X21.7
N44710 G01 X22.9
N44720 G01 X24.1
N44730 G01 X24.9
N44740 G00 Z5
N44750 G00 X6.9 Y5.9
N44760 G01 Z-7 F1200
N44770 G00 Z5
N44780 M02

ecl66, 2016-02-29:
[nounours.zip.zip](https://github.com/vilemnovak/blendercam/files/151157/nounours.zip.zip)

Hope this helps!
Cheers

vilemnovak, 2016-04-11:
Hello, in this case, the problem was too fine setting for optimisation, 
the reduction threshold in optimisation panel should be set higher, the default value is 0.2 um.
The problem is floating nubers precision, you precision was set to 0.00005 um, which is 0.00000005 mm, which is far beyond any machine precision, and the postprocessor doesn't export the differences which are actually there, otherwise it would remove the between points.
In future however, somebody might want to implement second layer of optimisation which looks forward in the gcode and writes only when values really change also in export, so I am not closing this issue, although it's not a real bug.

ecl66, 2016-04-11:
Hello,

Thanks a lot for your reply. I actually found out about this settings some times after posting - sorry about that.

What I am wondering now is how to set up a path based on the results from the previous path, such as finishing after roughing. So far, if I simply use two paths (for example a parallel path after roughing), there are plenty of useless paths that are milling into empty areas....

Keep up your great work please!

Best regards,

Harvie, 2017-05-10:
Not sure if this will help, but there are some g-code optimizers. for example:
https://github.com/alsliahona/gcode-optimizer
https://github.com/runeSal/OPTI-G

maybe blendercam can adopt some of their algorithms...

closed 2023-01-14 by pppalain