vilemduha/blendercam #34: Translations at Italian languages?

kylotron, 2016-03-18:
Hi all,
is possible to translate blendercam at other languages?

nfz, 2016-03-20:
Yes, its possible.  The plugin is not setup right now for other languages but it would not take much to do.

kylotron, 2016-03-22:
Ok... ;)

nfz, 2016-03-27:
Are you interested in providing the Italian translations?

kylotron, 2016-03-28:
Yes... :P

nfz, 2016-03-30:
Ok, I will start setting up support for International Languages support.  Will take a few days.

kylotron, 2016-03-30:
Beautiful. 
Thank you!

nfz, 2016-04-02:
Ok, was trying to use the Blender built in translation addon to setup a .po file for translations but encountered some issues.  This will take more time to get going.  I am investigating if its the Blender addon and/or blendercam that is causing the problem.

nfz, 2016-04-27:
Still working on this but hit a bit of a brick wall.  The blender addon works but it modifies the main blender translation file.  Only patches for official blender addon translations will get excepted into the main Blender git repo.  Looking at a different way of doing the translation similar to the way [bCNC](url) does it.

pppalain, 2023-01-14:
this enhancement was never finished

closed 2023-01-14 by pppalain