vilemduha/blendercam #36: Please add to the repository license for redistributable code

torvn77, 2016-03-23:
Asked me on one forum "Blender CAM as far as I understand a free product? As the Blender?"
At first I answered "Yes, all right."
And then I decided, and I will look what is written in the repository.
And there is nothing written, no license, and therefore to hold the product for use in the work.
Please add the license code, it would be better of course copyleft, the bourgeoisie was bad.
Original text:
Спросили меня на одном  форуме "Blender CAM насколько я понимаю бесплатный продукт? Так же как и сам Блендер?"
Сначала я ответил "Да, правильно понимаете."
А потом решил, а посмотрю я, что написано в репозитории.
А там ничего не написано, лицензии нету, а значит и брать продукт для использования в работе нельзя.
Добавьте пожалуйста лицензию на код, лучше бы конечно копилефт, чтобы буржуям плохо было.

nfz, 2016-03-24:
Good point.  I would assume GPL 2 as with all the other Blender addons but it should be stated in each file so that there is no guessing.  I can update the files if Vilem gives the go ahead.

nfz, 2016-03-24:
I should have looked before commenting.  __init__.py does state GPL 2.  I will put the GPL 2 header in the rest of the files.

closed 2016-03-24 by nfz

vilemnovak, 2016-03-25:
not sure if it was necessary, but it's good to be clear for everybody :) thanks.

torvn77, 2016-03-25:
Can the license file be added to the root of the repository?

nfz, 2016-03-26:
I can add a License file to the root.

torvn77, 2016-03-26:
Thanks, I wrote on this forum where was asked the appropriate question.

P. S.
Blender's license GPL3 :)

ksuprynowicz, 2016-03-27:
> Blender's license GPL3 :)

in fact it's GPL2 or newer, it allows more compatibility with other software:
https://www.blender.org/about/license/

commit 0ea9ae24 (2016-03-24): add GPL 2 license block

- fixes #36 issue - license unclear