vilemduha/blendercam #40: Unsafe start movement of toolpath

taras-ko, 2016-04-01:
Hello. I noticed that all kinds of generated toolpathes begin from the same fast movement.

1. It is rapid traverse to X0 Y0 Z0 location of current fixture:
G00 X0 Y0 Z0
2. Straight linear movement to first point for current toolpath
[image]

IMHO these are unsafe movements. You can damage your spindle on the 1 and 2 movements. I propose to split start position to three movements:

[image]
1. Go to the highest available Z coordinate of your machine.
For example the highest available Z movement on my machine is
     G53 G0 Z0
2. Go to the X0 Y0 of current fixture.
     G54 G0 X0 Y0
3. Go to the first point of current toolpath.

Thus we can be sure that our tool won't be cracked.

Could you please me help how to implement such feature. Just point me were to dig the code in the project. Thanks.

nfz, 2016-04-01:
I change the start position using options in the Cam Machine panel.  I enable Use Position Definitions and modify the start position ie set the Z axis start height at a safe height.  I also make use of Free movement height in the Cam Movement panel.

My cnc control software also has safety end limits, min height traverse, and speed checks enabled so that the gcode doesn't do silly things like do rapids at low heights and crash the tool into a clamp.

There is some info on the wiki but does need more work:
https://github.com/vilemnovak/blendercam/wiki/Blendercam-Panel-Descriptions#cam-machine-panel

taras-ko, 2016-04-03:
Thanks.

Here is a code snippet when I tuned "Start position" fields.

G00 X0 Y0 Z39.999
G01 X16.101 Y4.839 Z3 F500

First line is a  straight move to start position coordinates (which again could be unsafe movement) I personally dislike straight moves where x,y and z are involved together.
Second is straight move to the first toolpath point.
[image]

Again I would prefer to split this code to more safer.
Pseudo code:

1. go to safest highest Z of your machine (for my machine it's g53 g0 z0)
2. go to first point of toolpath X and Y.
3. go to first point of toolpath Z
[image]

Excuse me for repeating. Don't you think that this rules will help to human less bother with tool safety?

nfz, 2016-04-04:
Agreed.  Not hard to set up.  Rather than hard code it we could have an option for a gcode start preamble and an option to disable using the default start position.  The preamble could be anything you want since you would type in the gcode in the UI.

If you want to hack the code (hard code the stuff) for now to test out what you want then have a look at utils.py in scripts/addons/cam directory.  The function you want to modify is called exportGcodePath()  some where around line 1038 if using source from github.  exportGcodePath() calls the methods in the user selected post processor to output gcode to the file.  The parent post processor class is defined in nc.py located in scripts/addons/cam/nc directory.

taras-ko, 2016-04-05:
Thanks. How could I find the first point of toolpath?

nfz, 2016-04-05:
The toolpath is a list of vertices in a blender mesh.  The first vertex (0 index) is the start of the toolpath.  See line 1151 of utils.py and then look at 1213 to 1218.  That should give you a hint.

nfz, 2016-04-16:
With the new gcode header option in master branch you should be able to set this up now for your machine.

taras-ko, 2016-05-23:
Almost.

I need to add two lines to headrer
G53 G0 Z0
G53 G0 x0 y0

Is there a way to add multi line header/trailer?

taras-ko, 2016-05-23:
Well, for EMC2 NGC language there is a way to call subroutine from external file with a single line.
We can mark this request as 'DONE' if it's also appropriate for others.

nfz, 2016-05-25:
I use this and use multiline edit but maybe this is not in mainline blender.

nfz, 2016-05-25:
Looks like I missed when the blender team took out newline edit support in the StringProperty.  You use to be able to copy and paste from the text editor into a StringProperty ui object but now it only pastes in the first line.

I will search for other options to support multiple lines of text.  In the meantine, one quick solution that I could implement would be to use a comma `,` to mark a new line.  Here is an example:
`G53 G0 Z0, G53 G0 x0 y0`

taras-ko, 2016-05-25:
I suggest to use semicolon as a command splitter symbol. It's more native for SHELL scripts and C/C++ language.

nfz, 2016-05-25:
I agree, good call.

nfz, 2016-05-25:
I just updated the master branch to support multi line using ; as a line break in the gcode header and trailer.  Seems to work ok.  Let me know how it works for you.

taras-ko, 2016-05-29:
Ok, Thanks. It works!

taras-ko, 2016-05-31:
Mistakenly reopened.

closed 2016-05-31 by taras-ko