vilemduha/blendercam #41: Blender 2.7 bug in toolpath

TurBoss, 2016-04-01:
Hi i found a strange bug using blender 2.7 tool path in generation ( i know 2.7 not supported yet)
 better a pic

[image]

thx!

TurBoss, 2016-04-01:
here is the blend file if is useful 
https://drive.google.com/file/d/0B2zAzLRUwfiNNkJ3YmlLRy1JMFE/view?usp=sharing

nfz, 2016-04-02:
I'm using 2.77.  Will test your blend.

nfz, 2016-04-02:
Ok, confirmed.  This is actually an old problem I thought was fixed but obviously not.  When exact mode is enabled the Bullet physics engine is used for detecting collisions between the tool and the surface.  The problem is at startup the Bullet physics engine sets a default collision shape (a capsule) that should have been deleted or moved out of the way, which is what you are seeing in the generated tool path.

  The work around (until this bug is fixed) is to turn off "Use exact mode" in the Cam optimization panel.  This will use a height map image instead which will be much faster too in generating the tool path.

TurBoss, 2016-04-02:
thank you so much for your help :)

TurBoss, 2016-04-02:
also I cange the machine geometry to be under zero in x and y so if you moved de default shape arround I found it heheheh

vilemnovak, 2016-04-11:
Just fixed this, thanks for report!
It was an issue with scaling, the object was 100 meters away to not interfere, but I forgot that the whole scene gets scaled for simulation 100000 times, so those 100 meters made the cutter object being 1 cm from center :)

closed 2016-04-11 by vilemnovak