vilemduha/blendercam #42: Pack curves on sheet - not working

narvf, 2016-04-03:
Blender file attached.
I've used blender 2.69

Here is the output from console:

> Traceback (most recent call last):
>   File "/home/narvf/Pulpit/blendercam-master/scripts/addons/cam/ops.py", line 253, in execute
>     pack.packCurves()
>   File "/home/narvf/Pulpit/blendercam-master/scripts/addons/cam/pack.py", line 119, in packCurves
>     if allpoly.disjoint(ptrans):
> AttributeError: 'PreparedGeometry' object has no attribute 'disjoint'
> 
> location: <unknown location>:-1
>

nfz, 2016-04-04:
I get the same error with your blend in Blender 2.77 running Blendercam 9.1.  Definite bug.

nfz, 2016-04-04:
Found the problem.  I think Vilem was in the process of speeding things up by using packed geometry but its not complete.

But there is an easy fix to get it back to working but it will be slow again.  If you are up to modifying one line of code then you could have it working again.
Modify line 195 in scripts/addons/cam/pack.y from this:
`allpoly=prepared.prep(sgeometry.MultiPolygon(placedpolys))`
to this:
`allpoly=sgeometry.MultiPolygon(placedpolys)`

Just be warned that if you have a lot of curves it will be slow.

nfz, 2016-04-04:
I committed the temporary fix to github master if you want to pull from github for the fix.

nfz, 2016-04-04:
Took a crack at finishing the code to use prepared geometry.  It provides about 30% speed improvement on my machine over the old method.  Still room for improvement since its still slow.  I committed the changes to master if you want to try it out.

narvf, 2016-04-04:
Yes, now it is working! Thank you!
There is something like svgnest.com, it is also open source - I've discovered it today. Seems to work faster. Maybe source code from there would be useful.
https://github.com/Jack000/SVGnest

nfz, 2016-04-04:
Thanks for the tip, I will have a look at svgnest.com.

vilemnovak, 2016-04-05:
sorry guys, seeing this discussion just now.
that was really a stupid bug I didn't check when I was trying to use the prepared geometry all around blenderCAM. thanks for fixing, nfz! 
svgnest indeed looks promissing!

closed 2016-04-10 by nfz