vilemduha/blendercam #43: How to disable G43 Tool lenght offset command?

taras-ko, 2016-04-03:
Blendercam generates code
T1 M06 G43

I don't have an automatic tool change system. I'm changing tool in semi-automatic way by manually changing tool and probing it and calculating and applying offset difference using g code program. So for my machine

T1 M06 G43 is a bad command, because it applies offset for tool 1 from tool offsets table.

How about to add an option to disable tool management or better to give to user choice add a code snippet for auto tool changing sequence?

nfz, 2016-04-03:
Its generated by the post processor which is selected in the Cam Machine panel.  Which post processor are you using?

taras-ko, 2016-04-04:
emc2b I guess.

nfz, 2016-04-10:
The edge branch now has these features.
https://github.com/vilemnovak/blendercam/tree/edge

vilemnovak, 2016-04-11:
ok, closing this issue then :)

closed 2016-04-11 by vilemnovak