vilemduha/blendercam #45: [feature proposal] Adjustable speed of toolpath.

taras-ko, 2016-04-08:
Hi, folks.
I would like to propose use Edge Data "Crease" parameter to manually set the speed machine on "hard" places of toolpath.
This feature will be very useful on the beginning of every toolpath with deep plunge. A tool should mill the groove by cutting the material with all of its cutting surface, it's the most dangerouse circumstance of tool brokening. The next step will be much more easier for tool and we can increase the feedrate.
For example.
Suppose I have a toolpath which begins from deep plunge line, and supplose that the plunge depth and default feedrate are too tough for my tool. Ofcourse somehow we can implement an algorithm which will automatically detect such "hard" path regions and it will appropriatelly set the feedrate. But it's more easily give to the user ability to decide where he can slower/faster the feedrate.
[image]
I can quickly change 'Crease' parameter by pressing shift+e and moving the mouse. This data would be interpretted by postprocessor to change the feedrate
There is a problem. Crease parameter has range from 0.00 to 1.00. So we should think about formula how to set the feedrate slower/faster using such a range.

What are your suggestions about this enhancement?

nfz, 2016-04-10:
Interesting use of Crease.  

You can convert the crease value using the following formula to give a +1,-1 range:
`= (crease - 0.5) * 2`

taras-ko, 2016-04-10:
Even we can use crease 0.0-1.0 range as a coficient to override default feedrate. Say, defauld feed rate would be at 0.5 of a crease.

vilemnovak, 2016-04-11:
Hi, this feature is allready implemented for 3 axis machining, although it is experimental - you have to set Adjust feedrate with simulation in the Feedrate panel.
This creates a shape key which stores the computed speed adjustments. in the Z axis, you can display the feedrate adjustments when switching to the shape key on the path object.
This feature is however not well tested yet and could use some tweaking...

vilemnovak, 2016-04-11:
I forgot to write - it also exports the modified feedrates allready.
Edge crease however is also a good idea to visualise the speeds more naturally, however, with big paths, going to editmode might be very hard...

taras-ko, 2016-07-15:
I thinks we can close this issue it moved to issue #69

closed 2016-07-15 by taras-ko