vilemduha/blendercam #46: Windows 10 support shapely and geos_c.dll

TurBoss, 2016-04-09:
Hi i'm trying to install blendercam in a windows 10 machine

but when i load the addon in user preferences it drops the following error
`found bundled python: C:\PortableApps\Blender-2.77a\2.77\python
Traceback (most recent call last):
  File "C:\PortableApps\Blender-2.77a\2.77\scripts\modules\addon_utils.py", line 330, in enable
    mod = __import__(module_name)
  File "C:\PortableApps\BlenderCam\scripts\addons\cam\__init__.py", line 34, in <module>
    from cam import ui, ops, utils, simple, polygon_utils_cam#, post_processors
  File "C:\PortableApps\BlenderCam\scripts\addons\cam\ops.py", line 26, in <module>
    from cam import utils, pack,polygon_utils_cam,chunk,simple
  File "C:\PortableApps\BlenderCam\scripts\addons\cam\utils.py", line 41, in <module>
    from cam import chunk
  File "C:\PortableApps\BlenderCam\scripts\addons\cam\chunk.py", line 24, in <module>
    from shapely.geometry import polygon as spolygon
  File "C:\PortableApps\Blender-2.77a\2.77\python\lib\shapely\geometry\__init__.py", line 4, in <module>
    from .base import CAP_STYLE, JOIN_STYLE
  File "C:\PortableApps\Blender-2.77a\2.77\python\lib\shapely\geometry\base.py", line 9, in <module>
    from shapely.coords import CoordinateSequence
  File "C:\PortableApps\Blender-2.77a\2.77\python\lib\shapely\coords.py", line 8, in <module>
    from shapely.geos import lgeos
  File "C:\PortableApps\Blender-2.77a\2.77\python\lib\shapely\geos.py", line 114, in <module>
    _lgeos = CDLL("geos_c.dll")
  File "C:\PortableApps\Blender-2.77a\2.77\python\lib\ctypes\__init__.py", line 347, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] No se puede encontrar el módulo especificado

<!> event has invalid window`

any ideas?

is this a blendercam issue or is a shapely issue ??

thx

nfz, 2016-04-10:
A bit of both.  I think the package put together by Vilem for blendercam 0.9 is missing the geos_c.dll.  I don't know where you can get that missing dll without trying to do a full install of Shapely.

TurBoss, 2016-04-10:
I'm not a windows user too but i'm setting blendercam for a friend

the DLL is on the path

the same setup works fine in a windows 7 system

also the way I get shapely is from pip and copy and pase their folder to the blender bundled python (with dll too)

native python 3.5.1 works fine but blender bundled don't

not i a hurry just reporting this

thank you

marcel631, 2016-04-26:
I have the same experience. 32 bits blender cam.zip runs fine on Windows 7 but there is no way to get it running in Windows 10. Windows 10 is able to run a 32 bit program so it seems that there is some error in the python DLL loader. The error from Python is wrong because the DLL is found (and python states the module is not found). If one changes some bytes in the DLL, an exception is thrown. After that I Even tried to set-up blender in mini conda as was advertised as a solution but I could not handle it. I managed to install shapely in mini conda. I removed the python folder from blender 64 bits. Blender still starts so apparantly it finds my mini conda python. Then I put blender CAM add-on into blender and then import numpy fails. I do not understand why because numpy is installed in miniconda. In some way some path is apparantly incorrect. Thus this proves that there are simply too many paths and options involved. Cost me 2 days without result and now I give up. One would really need a youtube movie how to do this on Windows 10 (given that this mini conda solution would work on Windows 10).

marcel631, 2016-04-26:
OK I solved this. when you activate your environment in miniconda, the sys,path that belongs to your miniconda environment is not available in blender for some reason. If you print the sys.path you will see in a blender script just before import numpy. Thus mini-conda is useful for downloading a nice python folder but not for running. So lets say your miniconda environment folder is called "golem" then you will have a folder like c:\miniconda3\envs\golem. Next you copy all files inside golem to the blender python folder. Something like c:\blender\2.77\python. If it asks overwrite, say ok. Then finally copy all files, lot of exe and dll (thus not the folders) in the c:\blender\2.77\python to the c:\blender\2.77\python\bin folder. That did it for me in windows 10.to see the blender CAM add ons. To fill golem with the correct python files inside miniconde do this:

Install mini-conda voor python 3.5.1:
http://conda.pydata.org/miniconda.html 

launch command prompt: cmd (on Windows 10, press WIN + X then run, type CMD press enter)
at command prompt:
type "conda –n golem python=3.5.1" + enter
type "activate golem" + enter
test below should give python 3.5.1:
type"python –version" + enter
install shapely:
type "conda install -c ioos shapely=1.5.13" + enter

nfz, 2016-04-27:
This is really good info for the wiki.

TurBoss, 2016-04-27:
will try this at the moment that I have some spare time

ty

marcel631, 2016-04-29:
I tried to repeat the process on my laptop and decided to make a Youtube movie at the same time. There were all kind of new things happening but at the end it worked again and thus the film is 15 minutes. I also removed a few mistakes from my previous comment. When I installed the first time, I thought the shapely version was 1.5.13. But when I filmed the next day, it was 1.5.15. Maybe this is all changing very fast. Installation with miniconda is not always straight forward. Maybe it helps. See the post at:http://www.jointmaster.eu/2016/04/install-blender-cam/
OK, now I am going to try to speed up with blender cam and G-code.

TurBoss, 2016-05-04:
Worked like a charm

the only diference is that I only copied the shapely folder not all the other stuff

ty

closed 2016-09-11 by TurBoss

JohnRipa, 2016-12-04:
Seems basically the the geos module is incomplete but getting geos lib in binary for windows is a bit of a pain. I needed to download several things before I could build it. I am uploading it here in case it helps others.

You can try extracting this into:
blender-2.76-windows32\2.76\python\lib\site-packages\
[geos_win32.zip](https://github.com/vilemnovak/blendercam/files/629249/geos_win32.zip)

marcel631, 2022-09-18:
Could not get it working anymore using the comment above with blender 3.0, windows 10 64 bit with the latest blender CAM from github. Nowadays, blender comes with pip installed. You can run the script below from within a python console in blender. Save the script to a file "golem.py" in a temp folder and run the script in blender from the python console using:
"exec(open("c:/temp/golem.py").read())" + enter

You have to give "FULL control" in windows security settings for the blender folder. Otherwise it fails because of permissions.
Script might be handy also for people that want to use other modules in blender.

Script:

import subprocess
import sys
import os

# Script below installs shapely, opencamlib and equation in windows 10
# Make sure the security settings on the blender folder (right mouse button, properties tab) is "FULL control" for the user. 
# Save the content below in a file named c:\temp\golem.py
# Start python console in blender and type "exec(open("c:/temp/golem.py").read())" + enter
 
python_exe = os.path.join(sys.prefix, 'bin', 'python.exe')
target = os.path.join(sys.prefix, 'lib', 'site-packages')
print(target)
 
subprocess.call([python_exe, '-m', 'ensurepip'])
subprocess.call([python_exe, '-m', 'pip', 'install', '--upgrade', 'pip'])

#install (wheel):
from subprocess import PIPE, run

command = [python_exe, '-m', 'pip', 'install', '--upgrade', 'wheel', '-t', target]
result = run(command, stdout=PIPE, stderr=PIPE, universal_newlines=True)
print(result.returncode, result.stdout, result.stderr)

#install (shapely):
from subprocess import PIPE, run

command = [python_exe, '-m', 'pip', 'install', '--upgrade', 'shapely', '-t', target]
result = run(command, stdout=PIPE, stderr=PIPE, universal_newlines=True)
print(result.returncode, result.stdout, result.stderr)

 
from subprocess import PIPE, run

command = [python_exe, '-m', 'pip', 'install', '--upgrade', 'Equation', '-t', target]
result = run(command, stdout=PIPE, stderr=PIPE, universal_newlines=True)
print(result.returncode, result.stdout, result.stderr)

#install (opencamlib):
from subprocess import PIPE, run

command = [python_exe, '-m', 'pip', 'install', '--upgrade', 'ocl', '-t', target]
result = run(command, stdout=PIPE, stderr=PIPE, universal_newlines=True)
print(result.returncode, result.stdout, result.stderr)

#install (meta):
from subprocess import PIPE, run

command = [python_exe, '-m', 'pip', 'install', '--upgrade', 'meta', '-t', target]
result = run(command, stdout=PIPE, stderr=PIPE, universal_newlines=True)
print(result.returncode, result.stdout, result.stderr)

 
print('DONE')