vilemduha/blendercam #47: Edge branch

nfz, 2016-04-10:
I have set up an Edge branch (leading edge stuff) which will be for new features that will eventually be the next new release.  Since its leading edge it can be unstable at times ie crashes and things not working like they should.

https://github.com/vilemnovak/blendercam/tree/edge

I see the master branch as being the stable branch of the most recent release.  Commits to the master branch  would be mainly bug fixes until the next release.

So what is in the edge branch right now?  Here is a list of things that have been added since 0.9 release:

- pocket operation now supports "first down".  This means that if first down is enabled then each pocket will be completed (including all layers) before moving on to the next pocket.

- A gcode header and trailer can be enabled for each operation allowing you to put in your own gcode commands.  The validity of the commands is not checked so be careful.

- Block numbering can be turned on or off for most post processors.  You can also define the starting number and how much it increments by.

- Tool change commands can be enabled/disabled along with using G43.

- Medial axis operation (v-carve) now supports layers and first down.  If first down and layers are enabled then the carve stroke is completed to full depth before proceeding to the next stroke.

vilemnovak, 2016-04-11:
Thanks nfz, I will have to accomodate to this, since I always took releases as stable and what happened between was considered edge. As you know, there are many unfinished features in blenderCAM all the time...

vilemnovak, 2016-04-11:
Ok, I've been thinking about this... And I realized we allready have the Edge branch - it's the features which are marked EXPERIMENTAL. Could we keep it this way?
The thing is, if you never give to people new underdeveloped features, you never get the feedback you need to finish that stuff :) So experimental stuff from one release can become non-experimental in the next release?

nfz, 2016-04-11:
Yep, there are many ways of doing it.  I went with the edge branch because I had a backlog of additions and didn't want to throw them in the master branch since they were only tested by me.  I will look into how to set them up as EXPERIMENTAL.

closed 2016-04-11 by nfz

vilemnovak, 2016-04-11:
setting them up as experimental is pretty straightforward, you make an if statement on the experimental property, and also write it in the description. Also, If you still want to keep such a branch, I don't want to stop you from that ;) It's just my preferred way of doing this. 
Feel free to commit any stuff to master anytime ;)

nfz, 2016-04-11:
Looks straightforward.  I will do some more testing before committing to master.