vilemduha/blendercam #50: strategy for pockets

kylotron, 2016-04-25:
What strategy should I use to implement the attachment piece pockets 1,2,3,4 and 5?

The "piastra 2" operation is perfect!
The "piastra 1" operation, however, I can not calculate it ...

[image]

[pezzo.blend.zip](https://github.com/vilemnovak/blendercam/files/235047/pezzo.blend.zip)

nfz, 2016-04-27:
I can tell you how I do it but don't know if its the best way.  I use a curve object to define the outline of the pocket to be performed and place it at the depth needed to cut the pocket.  So for 1, 2, and 3 I would make 3 curve objects.  The curve object is used for the pocket operation instead of the original object.  With a single core  cpu running at 600 MHz I find this techique very fast for generating the g-code.

For areas 4 and 5 I would use parallel or block operation and use a limiting curve to define the area of operation.

nfz, 2016-04-27:
Here is an example:
[image]

I went into edit mode on the mesh, selected the edges that define the bottom of the pocket, Shift-D to copy, seperate - to make a new mesh. Then I select the new mesh and convert to curve and rename.  Now I can use that curve for the pocket operation.
[image]

nfz, 2016-04-27:
For area 4 I defined a limit curve just around the area I want to mill:
[image]

I setup a parallel operation using the model and enabled limit curve in the CAM Operation area panel:
[image]

kylotron, 2016-04-28:
oh yes, yes, yes.... 

thank's a lot

kylotron, 2016-04-29:
Pocket_1....

Why i have a most complicated path?

[image]

nfz, 2016-04-29:
What are the settings in the CAM optimization panel?  I didn't have OpenCAM lib enabled.

kylotron, 2016-04-29:
It's same!
[image]

[partiinpolicarbonato_1.blend.zip](https://github.com/vilemnovak/blendercam/files/243011/partiinpolicarbonato_1.blend.zip)

kylotron, 2016-04-29:
My curves for pockets are into other level... :)

nfz, 2016-04-29:
Just tried the zip you posted and I get the same thing as you.  Investigating....

nfz, 2016-04-29:
Ok, its a bug in the tool path generation code.  A work around for now is to use overlap of 0.8mm (instead of 1mm that you had) and it will work.

vilemnovak, 2016-04-30:
Hi, this is NOT a bug. 
Because of the big overlap, there would be uncut areas, so there need to be extra paths which finish those. Sorting of these extra paths is hard, because you need an hierarchy, which is very hard to define with more complex shapes. (any ideas here how to do this?)
Just use overcut =< radius of the tool.

vilemnovak, 2016-04-30:
Actually, most CAM softwares don't allow overlap larger than half of the cutter  - I was thinking why, only when I was developing the algorithm for detecting and cutting uncut areas, I realized why the coders were lazy to do such a complex thing :)

nfz, 2016-04-30:
I understand and maybe bug is a poor choice of words for this situation.

kylotron, 2016-05-02:
[image]

kylotron, 2016-05-03:
First plate finished!

[image]
[image]
[image]
[image]

in picture 2 you see a mistake the edge of the pocket because I had to reposition the piece to fall within the lower limit of the Z! :(

(sorry for my english)

kylotron, 2016-05-03:
Now there is a "limit" that should be passed ...
A limit curve which allows the processing only on the outside of the curve and not inside!

It's possible?

vilemnovak, 2016-05-03:
I don't understand what you mean now... there is a limit curve option. Can you explain better?

kylotron, 2016-05-03:
limit curve option allows RESTRICT operations **INSIDE** of the curve.
It would be useful to have a limit curve for RESTRICT operations **OUTSIDE** of the curve

vilemnovak, 2016-05-03:
Well, you can define limit curve from 2 curves inside each other, this way you can allready do this.

kylotron, 2016-05-03:
ok :D

Perfect! ;)

kylotron, 2016-05-04:
All photos for you!

[image]
[image]
[image]
[image]
[image]
[image]

closed 2016-05-29 by kylotron