vilemduha/blendercam #51: T-bone, Key, or other such inside corner strategy

dynamodan, 2016-04-25:
This feature may already exist, but I can't find it.  When doing profile cutout, what is the chance of being able to cut into the inside corners to avoid leaving a radius such as described here:

http://blog.inventables.com/2014/06/learn-about-milling-inside-corner.html

nfz, 2016-04-26:
I use one of the blendercam tools called overcut to modify the curve.  I then use the new curve generated fo the profile cutout.

There is a little blurb about it in the wiki:
https://github.com/vilemnovak/blendercam/wiki/Blendercam-Tools#add-overcuts

dynamodan, 2016-04-26:
That is excellent!  Thanks for pointing it out to me.  I had a feeling it was in there someplace!

vilemnovak, 2016-04-26:
Well, at least I know the interface should be fixed :)so that users actually see the features .

dynamodan, 2016-04-26:
The interface is good.  I should have read the wiki before asking. ;)  A few things to note:  1. Without "Outer polygons" checked, nothing seems to happen.  I'm not sure what this is supposed to do, beyond the obvious.  (The Invert checkbox works nice.)  2. The overcuts are always "dog-bone" and it would be nice for T-bone in cases of square tabbing, but maybe it's too difficult.  (I'm not especially good with geometric calculations or I would fork this myself, and add some settings to do this.)

nfz, 2016-04-26:
T-bone case can be added to the master branch.  I have a version working on my machine that allows T-bone orientated either on x or y axis.  I will generate a pull request for Vilem to review.

dynamodan, 2016-04-26:
This is exciting.  I'm going to reopen this issue for visibility as others may be interested in this too.

vilemnovak, 2016-04-26:
nfz - if you have it working just commit it ;) no need for pull requests.

nfz, 2016-04-27:
I decided to make the modified version that I use as a separate new overcut operator.  The main reason for this is it does work a little differently than the old one so didn't want to confuse people.

Here is what it looks like:
[image]
If there is only one curve then the operator will work even if "include outer curve" is disabled since there is only one curve to work with.  "include outer curve" is only useful when there are curves inside.
[image]

T-bone overcuts:
[image]

T-bone overcuts on the other edge with inverse:
[image]

T-bone overcuts on the other edge without inverse:
[image]

There is also using overcuts on opposite edges:
[image]

nfz, 2016-04-27:
I pushed this to my overcutb branch:
https://github.com/nfz/blendercam/tree/overcutb

vilemnovak, 2016-04-27:
Jeff, this looks very good to me, I don't have much time to go through the code now, but the results are just what this should look like ;)
So feel free to commit also to master.

vilemnovak, 2016-04-27:
also, when you already had work with those images, they could go to wiki too?

nfz, 2016-04-27:
The code didn't change much, I just re-organized it a bit.  The capability for T-bone was already there in the original code, it just had to be exposed ;).

When I update master I will update the wiki.

dynamodan, 2016-04-29:
Guys, this is fantastic.  It works perfectly on a tabbed "plywood cube" outline that I made using Addam's paper model export addon, that I forked, and added plywood-tab output:

https://github.com/dynamodan/Export-Paper-Model-from-Blender

Screenshot of T-Bone working:
[image]

Now for some more complicated, more angular stuff. ;)

nfz, 2016-04-29:
Cool, I am going to give your fork a try.  I need that for a project coming up, good timing :).

There are some issues with some shapes that I am still figuring out how to solve which is one of the reasons why this has not gone into master yet.  If you come across any strange results please post pics.

dynamodan, 2016-04-29:
Actually yes, I did discover an issue, but it's in a mode that I wasn't going to use at least for this particular object.  It could be a problem with shapes having odd angles besides 90.  It seems like the function is simply alternating, because when it encounters a condition where it overcuts an "odd" corner, the remaining corners are cut opposite, as in this screenshot.
[image]
It seems like the logic for determining overcut direction for T-Bone should look at the lengths of the lines that are either side of a given point, and choose to cut into the longer one rather than simply alternating. Not sure if that makes sense.

nfz, 2016-04-29:
You are correct in that the algorithm I used is simply alternating the edge when an inside corner is found and fails when there is an odd number.  Length alone does not always work either.  The two combined may cover most scenarios.  I am looking at some other cam software that have this functionality and learning how they deal with it.

nfz, 2016-04-29:
One method that I am trying right now counts turns instead of edges and seems to cover all the problem areas I have encountered.

nfz, 2016-04-29:
Yep, the turn method works really well.  I will update my github branch with this method.

nfz, 2016-04-30:
Nope, found more fail cases with the turn method.  Its better than the first two methods but there are some simple situations where it fails.  Back to the drawing board.

nfz, 2016-06-16:
Forgot about this branch until I needed it for a job.  Its now pushed to master.  There are still a few cases where T-bone is not correct and I need to put in some more rules to fix these when I figure them out :).

For most shapes and tabs, T-bone works but if you find a situation that does not work please post a screen shot.

closed 2016-06-18 by nfz