vilemduha/blendercam #53: Error... another... why?

kylotron, 2016-05-05:
`validity 
validity 
validity 
progress{building path pattern}
progress{0.6538865566253662}
convertViewVec: called in an invalid context
Traceback (most recent call last):
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 3284, in getPath
    getPath3axis(context,operation)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 2854, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 324, in sampleChunks
    getAmbient(o)  
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 1775, in getAmbient
    polys = curveToShapely(limit_curve)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 1364, in curveToShapely
    chunks=curveToChunks(cob)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/chunk.py", line 813, in curveToChunks
    co = meshFromCurve(o)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/chunk.py", line 793, in meshFromCurve
    co.data.dimensions='3D'
AttributeError: 'Mesh' object has no attribute 'dimensions'

location: <unknown location>:-1

location: <unknown location>:-1

`

Same file

kylotron, 2016-05-05:
with Ramp In enabled, have error.

nfz, 2016-05-07:
Can you post the blend file?

nfz, 2016-05-09:
ok, found the conditions where this happens.

vilemnovak, 2016-05-24:
was this fixed?

kylotron, 2016-05-25:
When i start Blender.... this is the first error:
[image]

...and when i move the mouse over panel at right of the screen, i have other same errors:
[image]

When i make a parallel strategy with Ramp In, this is the error:

[image]

nfz, 2016-05-25:
The UI AttributeError has been fixed in Master branch.

The other error in utils.py I also get sometimes when using ramp in.  Not sure what is causing it.

nfz, 2016-05-25:
Ok, found the problem.  The chunk method rampZigZag() does not return anything but a lot of code in utils.py expects it too.  Will provide a patch for Vilem to review.

nfz, 2016-05-26:
I committed some bug fixes to Master for ramp in so hopefully this fixes the problems you were having.

kylotron, 2016-05-26:
[image]

This is the result of Parallel strategy with Ramp In!

The path is not perfect... but is complite.

kylotron, 2016-05-27:
No Level...

[image]

[image]

closed 2016-05-29 by kylotron