vilemduha/blendercam #57: How to make this hole?

kylotron, 2016-05-16:
[image]

kylotron, 2016-05-18:
This is a countersunk hole (height 5 mm, max diameter 18.4 mm, min diameter 8.4 mm)... 
What strategy should I use? And how should I apply it?

Thank's

nfz, 2016-05-21:
Maybe circles or spiral using a limit curve?  I usually use a pocket operation for the straight vertical sided sections and then circles or spiral for sloped areas.

kylotron, 2016-05-22:
I fear it is not easy ...

nfz, 2016-05-22:
ok, giving it a try....

nfz, 2016-05-23:
Ok, you win.  I could not get a good path generated with any of the strategies.  Not sure what I am missing or why the strategies give such bad results but will continue investigating.

kylotron, 2016-05-23:
Ok.
In the meantime I'm writing a parametric gcode to bypass this problem.
But of course it is not "productive" to have to use different gcode to realize a single piece!
I look forward to your developments!

nfz, 2016-05-24:
With the latest changes that Vilem just committed, I can now use outlinefill strategy to generate a good tool path for this shape.  I did use a limit curve for the perimeter.

kylotron, 2016-05-24:
I receive this error!
[image]

kylotron, 2016-05-24:
[image]

nfz, 2016-05-25:
Are you using ramp in?

kylotron, 2016-05-25:
Not Ramp.In... Not Layers... Same Error!

My Zip file is not attached... :(

nfz, 2016-05-27:
I get that error if the limit curve is not a curve ie I forgot to convert the mesh to a curve.

kylotron, 2016-05-27:
The result is not perfect!

[image]

nfz, 2016-05-27:
Try it with **Use exact mode** enabled, I get much cleaner results.

kylotron, 2016-05-27:
Oh yessssssssssssssssssssssssss

[image]

closed 2016-05-29 by kylotron