vilemduha/blendercam #58: [feature request/discussion

taras-ko, 2016-05-19:


taras-ko, 2016-05-19:
I see than only one cpu is fully loaded during calculation of some huge toolpathes. I'm curious, is it really hard to load other cpu with calculation tasks, thus we can really speedup this, sometimes, headache process. Thanks.

nfz, 2016-05-21:
I have thought a lot about how to break up the calculation tasks into threads but so far have not found an easy and workable solution.  Hoping someone has some better insight.

taras-ko, 2016-05-22:
Ok. But we could give user an option to bind toolpath calculation to dedicated cpu. For example: bind first toolpath calculation to cpu0 and second toolpath calc to cpu1. And simultaneously calculate two separate pathes.

nfz, 2016-05-23:
Yes, that is an option.

vilemnovak, 2016-05-24:
This can be done with background operations allready. The only way to do multi threading is the same - run several blender instances in background and split the work. Why? because Python doesn't support really multicore, if things didn't change in the past two years or so.

ksuprynowicz, 2016-05-29:
The other way is to use BlenderCAM with OpenCAMLib. In such case toolpath sampling for most operations is offloaded to external C++ library. I've tested it with up to 24-core computers and it works perfectly. The only problem is that loading samples for multilayer operations needs optimization.