vilemduha/blendercam #59: The bundled i686 linux version is missing the libgeos library.

BMaxV, 2016-05-22:
Not a whole lot to say otherwise. I don't have root on that machine right now so my options are limited.

Thanks for this software though, I have another 64 bit computer I can generate files with.

nfz, 2016-05-28:
If you are getting blendercam from github then yes, you have to install Shapely, libgeos, and numpy.  On two machines I don't have root but I am still able to setup shapely and numpy in blender's bundled python site-packages directory.  libgeos used by Shapely is a little more challenging but it can be done with a local install.

m4rl0m, 2017-08-08:
If it is not too difficult can you explain here, broadly speaking also, how to install libgeos locally?

ksuprynowicz, 2017-08-08:
> If it is not too difficult can you explain here, broadly speaking also, how to install libgeos locally?

On Debian Linux, and probably Ubuntu, there's libgeos-3.5.1(or similar version) libgeos-dev and libgeos-c1v5 package. You can install them with aptitude, synaptic or apt. I suggest also installing all other dependencies and Blender from package manager, and then adding BlenderCAM to Blender script directory.

As a root:

    apt install libgeos-dev

m4rl0m, 2017-08-09:
Sorry, I meant without a root permission... 
I'm on Debian and have already downloaded libgeos-dev and his dependencies (libgeos-c1v5 and libgeos-3.5.1), but I don't know how to organize them on my home partition. I just need to know how to do it in general.
Thanks anyway so far!

ksuprynowicz, 2017-08-09:
I think that the you should extract this file from libgeos-3.5.1 into some local directory like $HOME/lib:

/usr/lib/x86_64-linux-gnu/libgeos-3.5.1.so

And add this directory to LD_LIBRARY_PATH. The problem is that some sources recommend running ldconfig after that, and it requires administrator privileges, but I think it's worth giving a try. You have to open terminal, and add directory with libgeos-3.5.1.so to LD_LIBRARY_PATH:

export LD_LIBRARY_PATH=$HOME/lib/:$LD_LIBRARY_PATH

and then from the same terminal run blender (LD_LIBRARY_PATH is set only for this terminal). In Blender you should enable BlenderCAM, and watch terminal for messages about missing libraries or other errors. Let me know if it works.

m4rl0m, 2017-08-11:
It doesn't works unfortunately, it keep asking libgeos_c or any of its variants just as you feared...
Thank you for having tried it anyway.

closed 2023-01-14 by pppalain