vilemduha/blendercam #64: Problem on parallel strategy

kylotron, 2016-05-30:
Friday works.
Today not work!

[image]

```
Traceback (most recent call last):
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 3324, in getPath
    getPath3axis(context,operation)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 2890, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 332, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 106, in oclSample
    exportModelsToSTL(operation)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 88, in exportModelsToSTL
    bpy.ops.export_mesh.stl(check_existing=True, filepath=file_name, filter_glob="*.stl", use_selection=True, ascii=False, use_mesh_modifiers=True, axis_forward='Y', axis_up='Z', global_scale=1.0)
  File "/home/kylotron/Scaricati/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/modules/bpy/ops.py", line 189, in __call__
    ret = op_call(self.idname_py(), None, kw)
TypeError: Converting py args to operator properties: : keyword "use_selection" unrecognized

location: <unknown location>:-1

bpy.context.space_data.recent_folders_active = 2

```

kylotron, 2016-05-30:
No... ok. It's work.:(

kylotron, 2016-05-30:
Please verify "exact mode" with Parallel .

kylotron, 2016-05-30:
[image]

nfz, 2016-06-02:
Are you still having a problem with Use exact mode + OpenCAMlib?

kylotron, 2016-06-03:
Yes... :(

This is the first test....

[image]
[image]
[image]
[image]

I made a new working process with parallel strategy, "exact mode" and OpenCAMlib library activated. This is the error generated:

`progress{processing curve 25.0%}
progress{processing curve 50.0%}
Traceback (most recent call last):
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 3329, in getPath
    getPath3axis(context,operation)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 2895, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/utils.py", line 332, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 106, in oclSample
    exportModelsToSTL(operation)
  File "/home/kylotron/Scaricati/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 88, in exportModelsToSTL
    bpy.ops.export_mesh.stl(check_existing=True, filepath=file_name, filter_glob="*.stl", use_selection=True, ascii=False, use_mesh_modifiers=True, axis_forward='Y', axis_up='Z', global_scale=1.0)
  File "/home/kylotron/Scaricati/blender-2.76b-linux-glibc211-x86_64/2.76/scripts/modules/bpy/ops.py", line 189, in __call__
    ret = op_call(self.idname_py(), None, kw)
TypeError: Converting py args to operator properties: : keyword "use_selection" unrecognized

location: <unknown location>:-1

location: <unknown location>:-1
`

This is the second test

[image]

kylotron, 2016-06-03:
Another test...

1) Parallel + angle = 90° + Exact Mode + OpenCAMLib + Ramp In ----------- (no Use Layers)
[image]

2) Parallel + angle = 90° + Exact Mode + OpenCAMLib  ----------- (no Ramp In) (no Use Layers)
[image]

3) Parallel + angle = 90° + Exact Mode ----------- (no OpenCAMLib) (no Ramp In) (no Use Layers)
[image]

4) Parallel + angle = 90° ----------- (no Exact Mode) (no OpenCAMLib) (no Ramp In) (no Use Layers)
[image]

5) Parallel + angle = 90° + OpenCAMLib ----------- (no Exact Mode) (no Ramp In) (no Use Layers)
[image]

6) Parallel + angle = 90° + OpenCAMLib + Ramp In ----------- (no Exact Mode) (no Use Layers)
[image]

nfz, 2016-06-06:
I am not able to re-create this. Could you post the blend file here?  You can drag and drop the zip into the message area.

kylotron, 2016-06-06:
This is by test file
[test.blend.zip](https://github.com/vilemnovak/blendercam/files/300028/test.blend.zip)

nfz, 2016-06-06:
Thanks for the file.
I now get the same as what you get.

ksuprynowicz, 2016-07-10:
Hi!
I've just checked the file you attached and it works perfectly with:
Exact Mode + OpenCAMLib ----------- (no Ramp In) and Blender 2.77a
(OpenCAMLib is only used when both 'Exact Mode' and 'use OpenCAMLib' are active)
The error you are getting is due to recent Blender API change. Try using version 2.77a or newer.
When Ramp In is activated, there's always empty trajectory, no matter if OpenCAMLib is active or not.

closed 2023-01-14 by pppalain