vilemduha/blendercam #65: No CAM toolbar in linux remote desktop connection with blender-softwaregl

levasc, 2016-05-30:
Hi, blender CAM 0.90, ubuntu, the CAM toolbar not displayed when started with blender-softwaregl (it wont start with blender in remote desktop connection). 
Here what I see after downloading blender CAM:
https://plik.ovh/file/9tHLlLnp8GGHocir/KCqPDsOfofIsKk6n/Screenshot%20from%202016-05-30%2017:53:39.png

nfz, 2016-05-30:
The is some info on the wiki that might help:
https://github.com/vilemnovak/blendercam/wiki/Using-Blendercam-from-github

Make sure you have Shapely and libgeos installed.  If you are using a blender version that has its own bundled python then the dependencies have to be installed to the site-packages directory of the bundled python.

Have a terminal window open for blender to see any error messages.

levasc, 2016-05-31:
Thank you! Installing  shapely and libgeos solved the issue, now the cam toolbar is available, and I can use Blender CAM with public cloud services.

closed 2016-05-31 by levasc