vilemduha/blendercam #66: [UI improvement proposal] Cam toolpath mesh should be automatically selected on toolpath selection in CAM operations window

taras-ko, 2016-05-31:
Thus we can easily navigate through the many toolpathes, and won't be confused.

nfz, 2016-05-31:
+1.  I could start this in a couple of days after I get some cutting done.

nfz, 2016-06-02:
Started working on this.

If there is no cam path generated then when the operation is selected, the object/group is then selected.

nfz, 2016-06-04:
This is now in master.

taras-ko, 2016-06-05:
I found that creating relationship beetween object (Parent) and its cam toolpath mesh (children) is really convenient thing. By [parenting](https://www.blender.org/manual/editors/3dview/object/relationships/parents.html) objects and its toolpathes we can really organize our outliner. Now cam toolpath is folding in as a subnode of object in outliner. Besides 'children' toolpath is synchronously moving with its 'parent' object.

nfz, 2016-06-06:
+1
That is a really good idea.  Yep, easy to do and will implement.

nfz, 2016-06-06:
First implementation of parenting is in master.  For now this only works if source of data is an object.  Does not work if source of data is a group or an image.

nfz, 2016-06-08:
Added some more 3d view selection enhancements.  If the source of data changes or object/group changes then the selection will update.

taras-ko, 2016-06-09:
Cool, I'll check it.

nfz, 2016-06-11:
The last thing to be implemented is highlight selection for Chains.  Not sure if it would be beneficial or just cause confusion.

closed 2016-06-18 by nfz