vilemduha/blendercam #67: [UI extension proposal] Use not only group of objects for bridge generation

taras-ko, 2016-06-02:
Bridges also could be generated from 1 curve object with attached modifiers to it.

Use case:
Very often, I'm creating bridges using one curve with attached array and mirror modifiers to it. Thus I have one quickly managed object which represents all bridges. Then, I'm converting it to mesh (modifiers automatically are applying), than again converting it back to curve. At last I have my bridges, but they are a single object. So I forced to create a group based on that single object, and only then I proceed with bridge generation.

So I would be very good if we would have a bridge generation based on one curve object with an attached modifiers to it.

nfz, 2016-06-02:
+1.  Good idea.

nfz, 2016-06-04:
Started working on this.

Do you have a blend as an example I could use for testing?

nfz, 2016-06-04:
I've put a property in the bridges ui for "Use Modifiers".  if it is enabled then modifiers are applied and the result used as bridges.  Seems to work with my simple test example.

taras-ko, 2016-06-04:
Ok. Here is the sample [file](https://cloud.openmailbox.org/index.php/s/6jnGPp21xa5zHzF). Its size is 2Mb, but when I'm applying modifiers it became huge. I prepared two operations. First is a parallel milling on object with attached three modifiers: subsurve, array and curve. Second the same but it's a curve and they are bridges.
Please try it on your code.

nfz, 2016-06-05:
That is perfect.  Seems to work with no problems.  I will try and get the code for this in master tomorrow.

nfz, 2016-06-05:
Actually, there was a problem with the profile cutout and use mesh modifiers that I missed before.  My tests were too simple.

nfz, 2016-06-05:
Found a problem with ambient calculation when using modifiers.  Will have to wait until tomorrow to get fixed.

nfz, 2016-06-05:
Ok, I think all fixes are in master now.  The only change I did to your file was add the bridge curve to a group called BridgeGroup (you can call it anything) and set that group name in the Bridges group property. Enable use modifiers for the mesh and the bridges and all should work.

taras-ko, 2016-06-05:
Well, I tried to build path without no success. I noticed that blender process eate all my RAM (8GB) than it eate all my swap (4GB), after that my whole system has been frozen till warm reset. I tried two times in a row, and got system hangup after total RAM and swap occupation by the /usr/bin/blender app.

taras-ko, 2016-06-05:
The second cam toolpath called 'profile' is generating pretty fast and stable.

nfz, 2016-06-05:
hmm, odd.  I only have 800 megs of ram and didn't have a problem with "1".  I will recheck my local copy to see if I forgot to push something to Github.

nfz, 2016-06-05:
ok, found the problem, I think.  When  "use exact mode" is enabled then I do have a problem with running out of ram.  When "use exact mode" is disabled ie image mode is used, then there are no problems on my machine.  The bullet collision object that gets generated from the mesh seems to be the culprit.  Trying to find a solution.

nfz, 2016-06-05:
My bad.  Forgot to remove the modifiers in the object copy after applying the modifiers to the mesh.

nfz, 2016-06-06:
Fix is in master.  I had no problem with "use exact mode" and it was much faster than image method.  I wasn't happy with the parallel path generated but I think the angle needs to be adjusted, maybe 90 deg would be better.

nfz, 2016-06-11:
So far I have not run into any issues since the last fix with this feature but I don't do a lot of fancy cutting.  I have been using subdivision, arrays, bevel, and boolean, with no problems.  If no one is having problems with this then I will close it.

closed 2016-06-18 by nfz