vilemduha/blendercam #71: heidenhain post

TurBoss, 2016-06-27:
any one working on it?

nfz, 2016-06-27:
There is a post processor for heidenhain controller but it has not been tested.  I can modify it, or you can, if you have problems with it.  Its hm50.py in the nc subdirectory.

vilemnovak, 2016-06-27:
I guess the postpro is originally from the nc library, so it should work since I think the developer had some heidenhain machine.

TurBoss, 2016-06-27:
i thin that the file is heiden.py

and outputs
```
10BEGIN PGM 0 MM
11G21
(G-code generated with BlenderCAM and NC library)
12G17G90
(Tool change - D = 2.0 mm  type BALL flutes 2)
13T1M06
S12000M0314G00X199999.984Y200000Z9999999.023
15G01X148449.218Y181546.585Z9999.999F600
16G01Z-50999.996
17G01X148024.932Y181970.87F2000
18G01X148053.222Y181772.872
19G01X148060.302Y181723.388
20G01X148201.721Y181581.954
```
and should be 
```
#1 BEGIN PGM 0011 MM
#2 BLK FORM 0.1 Z X-262.532 Y-262.55 Z-75.95
#3 BLK FORM 0.2 X262.532 Y262.55 Z0.05
#4 TOOL CALL 3 Z S3263 DL+0.0 DR+0.0
#5 TOOL CALL 3 Z S3263 DL+0.0 DR+0.0
#6 L X-80.644 Y-95.2 Z+100.0 R0 F237 M3
#7 L Z-23.222 F333
#8 L X-80.627 Y-95.208 Z-23.5 F326
#49 L X-73.218 Y-88.104 Z-26.747 F229
#50 L X-73.529 Y-87.795 Z-26.769 F227
#51 L X-74.09 Y-87.326 Z-25.996 F279
#52 M30
#53 END PGM 0011 MM
```

nfz, 2016-06-27:
Sorry, it is heiden.py.  Read the wrong line in utils.py.

nfz, 2016-06-27:
From what I read on the Heekscnc forum from a few years ago it was a user that submitted the post processor but never got it working correctly.

Shouldn't be too hard to correct this.

TurBoss, 2016-07-12:
I started writing a new one only left fix the header curves and drills

TurBoss, 2016-07-17:
I continue this on #72

closed 2016-07-17 by TurBoss