vilemduha/blendercam #73: Issue with OpenCAMLib guide.

taras-ko, 2016-07-02:
Hi. Tried to use [this](https://github.com/vilemnovak/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib) mini-guide and get failed.
[image]
Any ideas? Thanks.

ksuprynowicz, 2016-07-10:
Hi!
There was a change in parameters accepted by Blender's STL export function recently and I've updated OpenCAMLib interface to be compatible with it. Can you try with more recent Blender version, like 2.77a?

taras-ko, 2016-07-14:
Yeap. It works with 2.77a. Unfortunatelly there are no support for modifiers, disscussed ealier in #63 
Also I need to know what strategies OpenCamLib is supporting in blendercam property panel.
Thanks.

taras-ko, 2016-07-14:
Also I found "Outline fill" strategy is very useful for my projects. But Its calculaition time terribly long. I'm curious wether OpenCamLib supports calculation of such strategy. Thanks.

ksuprynowicz, 2016-07-15:
It partially does. You have to select both "use exact mode" and "use OpenCAMLib". The outline is still computed and offsetted by BlenderCAM, but path sampling(the most time consuming part) is then done by OpenCAMLib. I've just tested it on old 24-core machine, and it's way faster than original sampling algorithm. It also works with 'Carve' and some other operations. Let me know if it works for you.

ksuprynowicz, 2016-07-15:
And as for modifiers, I'll have a look today, I think that there's a simple fix for it.

pppalain, 2023-01-14:
opencamlib now autoinstalls along with other dependencies

closed 2023-01-14 by pppalain