vilemduha/blendercam #74: starting point of gcode

TurBoss, 2016-07-03:
Hi

how i can define the starting point fo the gcode 
eg= x0 y0 z50 ?

http://imgur.com/P0T6Wd7

taras-ko, 2016-07-05:
[image]
Maybe you need g-code header option.
In this example. My machine goes to the safest Z point. (for my machine safest point in machine coordinates is G53 Z0) After that it goes to the x0 y0 of G54 offset, or whatever you'll point. Symbol ';' is like a newline character.

TurBoss, 2016-07-06:
In my case the piece is independent of the machine coordinates
I place it with g54 on the machine's control panel and the "0" of the piece is the top and center X and Y

thx for your answer

closed 2016-07-12 by TurBoss