vilemduha/blendercam #75: [UI improvement proposals] Distance between toolpathes and automatic visibility toggling of toolpathes.

taras-ko, 2016-07-05:
1. Distance between toolpathes should be measure in percents of current cutting tool diameter. Default distance should be 40% of cutting tool diameter.

2. By selecting of each generated toolpath other toolpathes should be hidden.

ksuprynowicz, 2016-07-10:
1. I think that both distance in millimeters and tool diameter percentage is reasonable, but I personally prefer having distance in millimeters - it's done this way in a lot of commercial software and it's easier to estimate surface roughness after parallel pass. On the other hand if methods like trochoidal machining were implemented, percentage would be a good idea for them(it's directly related to cutter load).
2. I strongly disagree with this - while automatically selecting toolpath when the operation is selected is a good idea, hiding other toolpaths automatically could lead to very expensive mistakes. Frequently a single feature is done with multiple operations, for example a roughing pass and two finishing passes for a pocket(walls and bottom). Seeing them all at once is required to prevent undercutting and breaking cutter on finishing pass(Or even damaging spindle when using very large indexable cutters). While a lot of toolpaths visible at once can be unreadable, they can be easily and selectively hidden from outliner.

taras-ko, 2016-07-15:
> Frequently a single feature is done with multiple operations, for example a roughing pass and two finishing passes for a pocket(walls and bottom). Seeing them all at once is required to prevent [...]

This feature is can be nicely seen in your CNC controller graph window, I guess. For example: I use LinuxCNC and it draws a trace line for previous toolpathes. It helps me do not lost in number of toolpathes.

ksuprynowicz, 2016-07-15:
For some controllers and small programs it can be done this way, but for larger ones (several gigabytes, 20 or more hours on fast CNC machine) there's no controller that can show this in practical way. And in such cases costs in lost material can easily run into thousands of dollars.

taras-ko, 2016-07-16:
> I strongly disagree with this - while automatically selecting toolpath when the operation is selected is a good idea, hiding other toolpaths automatically could lead to very expensive mistakes.
Back to the discussion again. I think, at least, making the toolpath automatically visible (if it's hidden) on tool operation selection improve usability of Blendercam. How do you think?

ksuprynowicz, 2016-07-16:
> Back to the discussion again. I think, at least, making the toolpath automatically visible (if it's hidden) on tool operation selection will improve usability of Blendercam. How do you think?

Definitely yes :) Automatic selection would be also practical.

ksuprynowicz, 2016-07-16:
As I think of it, automatic hiding of toolpaths would also be very useful (sometimes scene gets cluttered horribly), but it should be available as an option/checkbox.

ksuprynowicz, 2016-07-16:
I didn't look much at BlenderCAM GUI code(only to add OpenCAMLib checkbox), but I think that there's an event that happens when operation is selected - code for your suggestion could be added there.

taras-ko, 2016-07-16:
Ok. I'll try to dig in the code.

taras-ko, 2016-07-19:
> As I thought more about it, automatic hiding of toolpaths would also be very useful (sometimes scene gets cluttered horribly), but it should be available as an option/checkbox.

I've just added a pull request with this feature that we disscussed, could you please verify? It's #81

jeffmd, 2016-07-19:
I have % tool diameter for distance between toolpaths in my own branch along with actual distance.  You can change one and the other will update automatically.  Not sure if that is what you are looking for.
[image]

taras-ko, 2016-07-19:
> I have % tool diameter for distance between toolpaths in my own branch along with actual distance. You can change one and the other will update automatically. Not sure if that is what you are looking for.
Exactly! It would be great to have it in master branch.

jeffmd, 2016-07-19:
I don't have write access to this repo anymore so will do a pull request.

ksuprynowicz, 2016-07-19:
> I have % tool diameter for distance between toolpaths in my own branch along with actual distance. You can change one and the other will update automatically. Not sure if that is what you are looking for.

Thanks! That's the best way of doing it :)

taras-ko, 2016-07-19:
Back to the last item
> Add possibility to use signle objects for bridge generation.

What do you think on that?

jeffmd, 2016-07-19:
I modified your current pull request, which just got applied to master, to allow object selection if operation object selection enabled in the UI settings panel.  Also modified your code so its not per operation ie hide all other paths, its now in the UI settings panel.  Still playing with the layout of the UI settings panel and will post a pic when done.

ksuprynowicz, 2016-07-20:
> What do you think about this?

It would be very practical.

closed 2023-01-14 by pppalain