vilemduha/blendercam #76: Email notification ad attachments?

TurBoss, 2016-07-06:
Hi

can the plugin have some trigger  to send an email or a telegram with another plugin when when the job has been calculated

I have no idea how to acomplish this

it tooks a while so I can't go to the computer every 5 mins to check if is done

this is just a suggestion thx!

taras-ko, 2016-07-15:
Yeah. Blendercam calculation power is limited to one cpu core, unfortunately. Try to use calculation in background, and then you'll see calculation progress bar.

I think, that Blendercam need to activate a timer each path calculation to register time elapsed during calculation. Next calculation cycle it can print approximate time, based on the previous time measuring.

closed 2016-07-16 by TurBoss

taras-ko, 2016-07-21:
Hey, I've just remembered. Try to launch Blender from within console, then you'll see progress percentage in console output