vilemduha/blendercam #77: Aluminium

TurBoss, 2016-07-06:
I'm goin to start testing blender cam on aluminium

any tip or precautions?

thx

ksuprynowicz, 2016-07-08:
Hi!
I machine a lot of aluminum commercially with Blender CAM, and things that I can suggest are:
- before machining watch and simulate trajectory with CAMotics (http://camotics.org/). This is a voxel-based simulation, so when you set the cutter length correctly, you can check for undercuts.
- for first machining try AW-2017 alloy - it has very good machinability and doesn't build up on cutter as much as for example AW-1050. Some alloys can be very problematic.
- use climb cutting
- adding oil mist or MQL(minimum quantity lubrication) will greatly extend tool life
If you have any questions, feel free to ask :)

ksuprynowicz, 2016-07-08:
Also, you should use mills made for machining aluminum. I recommend single flute carbide mills.

TurBoss, 2016-07-08:
Thank you so much

ksuprynowicz, 2016-07-09:
You're welcome :)
One more thing:
When machining pocket it's best to do it in two passes:
For roughing pass increase cutter diameter in BlenderCAM, so for example you enter 6.5mm for 6 mm cutter. This leaves 0.25 mm of material on pocket's sides. Then in finishing pass use 'cutout' operation and enter real cutter diameter (i.e. 6mm) and greater layer depth. This way you can get shorter machining time and better surface quality of pocket's sides.

TurBoss, 2016-07-09:
the machine is working now it will every thing is fine
I will only machine surface with texture like this http://imgur.com/P0T6Wd7
I'm using a Heidenhain control so the pockets are done from the nc panel itself but thx anyway

closed 2016-07-09 by TurBoss