vilemduha/blendercam #79: error installing OpenCamLib Ubuntu 16.04

TurBoss, 2016-07-17:
hi

I have build ocl from sources with vtk6 support but I made a deb package
once installed I try to import ocl from python but I get this:

```
$ python3
Python 3.5.1+ (default, Mar 30 2016, 22:46:26) 
[GCC 5.3.1 20160330] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ocl
Violación de segmento (`core' generado)
```

seg fault

ksuprynowicz, 2016-07-17:
It's because OpenCAMLib should be built with python 2.7. You can change that through cmake-gui interface.

TurBoss, 2016-07-17:
Ok thx Ive installed opencamlib correctly now

but when i go to blender and click the calculate button i get this:

```
Traceback (most recent call last):
  File "/home/jauria/BlenderCAM/blendercam/scripts/addons/cam/ops.py", line 218, in execute
    utils.getPath(context,o)
  File "/home/jauria/BlenderCAM/blendercam/scripts/addons/cam/utils.py", line 3347, in getPath
    getPath3axis(context,operation)
  File "/home/jauria/BlenderCAM/blendercam/scripts/addons/cam/utils.py", line 2913, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/jauria/BlenderCAM/blendercam/scripts/addons/cam/utils.py", line 340, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/jauria/BlenderCAM/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 111, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/jauria/BlenderCAM/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 48, in chunkPointSamplesFromCSV
    csv_file = open(tempfile.gettempdir()+'/ocl_chunk_samples.txt','r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: /home/jauria/BlenderCAM/blender-2.77/2.77/scripts/modules/bpy/ops.py:189
```

any ideas?

thx

taras-ko, 2016-07-17:
I think your python cant import some other modules. Please add a full console output.

TurBoss, 2016-07-17:
here is the full output
http://dpaste.com/2TMSW7P

taras-ko, 2016-07-17:
Hm, it's hard to say what's happening. Please share your file, I'll try to calculate in on my desktop.

TurBoss, 2016-07-17:
will try to reproduce in a new file

TurBoss, 2016-07-17:
any new file that I created has this issue :(

this is a new file with the same issue https://mega.nz/#!QF5jBYhZ

ksuprynowicz, 2016-07-17:
I'd check this out, but I can't download the file you provided - mega.nz site asks for a decryption link.
The weird thing about the console output is that OpenCAMLib script doesn't seem to be called at all.
What happens when you execute system-wide python 2.7 interpreter and type:

   import ocl

TurBoss, 2016-07-17:
the file is a new plane with random sculpt (any heightmap could do the same)

the output is :

```
$ python
Python 2.7.12 (default, Jul  1 2016, 15:12:24) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ocl
>>> 
```

TurBoss, 2016-07-17:
the link fixed : https://mega.nz/#!BZJDxQqD!mfkERhmMboiJJ1q-70UuN4cKCVSB1NnuFKm0L3lCtFs

ksuprynowicz, 2016-07-18:
There are two problems with this file:
You have to set operation depth start in "cam movement" to 1cm, as you use positive coordinates, otherwise no toolpath will appear.
For ball cutters you have to change cutter type from "sphere" to "ballnose"(toolpath won't be computed at all without this)
Let me know if this solves the problem.(It worked on my computer)

TurBoss, 2016-07-18:
Ok it worked and very fast

but i have a question : the path didn't go up in Z so the surface hasn’t been copied
eg:
http://i.imgur.com/qJhkqcw.png
not all the surface has paths

ksuprynowicz, 2016-07-18:
Check if you have "operation depth start" in "cam movement" set to 1cm. It's 0m by default, so when you use positive z coortdinates, you'll get exactly this.

TurBoss, 2016-07-19:
will test ASAP because linux computer hungs under high load
thx and sry

closed 2016-09-11 by TurBoss