vilemduha/blendercam #80: can we have gitter?

TurBoss, 2016-07-18:
a chat solution for github repos
as irc is to big for the people we are and issues for questions is the plan

https://gitter.im/

thx

taras-ko, 2016-07-18:
I can create a chat room only for my fork ot blendercam. We need to ask vilem to create a chatroom for his repo.
@vilemnovak  Could you please create a chat room for your project? Thanks.

vilemnovak, 2016-07-18:
There is a blendercam forum. How to create a chat room? Sorry if there is a way to give you rights i'd like to do it since I am completely out of time.

taras-ko, 2016-07-18:
Unfortunately I can't create a chat room for vilemnovak/blender. I'm not the owner.

taras-ko, 2016-07-21:
+1 I think we need a chat room with possibility to share the screen.

closed 2016-09-11 by TurBoss