vilemduha/blendercam #83: [Improvement proposal] Approximation to arcs

taras-ko, 2016-07-21:
It would be great if we will have this feature.
On the video you can see these jerky movements.

https://youtu.be/vB_sBSFH_1A

Agree that smooth movements are crucial for milling the aluminium.
The most improvement that with arcs approximation we will be able to speed up execution of a program, because each of these short toolpath pieces (edges) requires acceleration and deceleration in the begin and at the end of each piece.
Also we will have more compact toolpath files and will save the lifetime of mechanical parts wearing out of machine.

Thanks.

ksuprynowicz, 2016-07-26:
Hi!
This is definitely problematic, and as you said will quickly wear out both cutter and machine. The problem is that as far as I know, there's no easy way to add arcs to BlenderCAM(please correct me if I'm wrong).
The easiest way to improve smoothness of tool motion is to sample toolpath more frequently.
Far better solution is to interpolate finished trajectory with splines. I've written a simple program in C++ that detects sharp edges and interpolates the rest of the path with Catmull-Rom splines. Sharp edge threshold can be set in the code and edges are preserved this way. This improves machining times, surface quality and tool life. When using indexable carbide 16mm cutter in AW2017 aluminum alloy with 3m/min feed and 18000rpm spindle speed with MQL lubrication, with spline interpolation i get tool life of more than 30 hours.
The program typically outputs trajectory as position in time(it takes acceleration into account), but can trivially be converted to output gcode. If you are interested in it, I can put it on Github.

vilemnovak, 2016-07-28:
This is also up to the machine settings, good machine control should be able to interpolate on the fly. 
Arc detection is quite possible, I had a piece of code for that lying around. There is however a BIG problem with arcs and 3d paths - they are not possible together :) at least as far as I know. Arcs are only basically for 2d. Or am I wrong Karol? maybe you know more.. The fact I realized arc gcodes are only for XY, XZ, and YZ planes made me not to implement this at all in the end.

ksuprynowicz, 2016-07-28:
You are completely right :)
On my CNC controller path is converted into 3d splines upon loading it. In my opinion it's more practical, because only some parts of the toolpath could be converted to arcs, and whole path can be converted to spline segments with sharp edges only where necessary. When path vertices lie on arc, it's interpolated prefectly. The same happens for ellipses and 3d shapes like spirals. That would be impossible with gcode arcs.
Most machine controllers do it in similiar way, there's typically setting for trajectory tolerance, so path points lying on the arc will get interpolated into arc trajectory anyway.
For example in LinuxCNC there's G61, G61.1 Exact Path Mode command for no interpolation and and G64 Path Blending for smoothing the path. It accepts tolerance as a parameter.

taras-ko, 2016-07-31:
> For example in LinuxCNC there's G61, G61.1 Exact Path Mode 

Thanks for the tip. I'll test it how it will fit for my needs.

ksuprynowicz, 2016-07-31:
> Thanks for the tip. I'll test it how it will fit for my needs.

Here you can find more info:
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?TrajectoryControl
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G61-G61_1
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G64

taras-ko, 2016-08-19:
@ksuprynowicz 
Hi.
I tested this trajectory planner feature
[image]
Yep. G64 P0.1 command will do the job. But as you can it's not very accurate blending. Trajectory is overcutting target model. A think, arc approximation will be more precise and fast as well. It's not very important issue, but it would be great to have it in the future, besides all modern proprietary CAMs have this feature.

closed 2016-08-19 by taras-ko

ksuprynowicz, 2016-08-21:
Hi!
Can you tell me what type of operation is it(parallel, waterline, pocket, etc.)?
Generally to achieve precision with toolpath blending you have to set blending tolerance value far lower, like 0.02mm or even 0.01mm, and you have to generate very high quality toolpath (for example for parallel I use 0.02mm distance between sampling points).
Try setting blending tolerance to  0.02mm or 0.01mm with high quality toolpath and if this won't help, then the best solution would be to interpolate and smooth path with splines. Arcs are good only for some situations, splines are ok for all the cases.
If you're interested, I can upload good code for intepolating toolpaths with Catmull-Rom Splines and with edge detection to github. It goes EXACTLY through each point in toolpath, but smooths it in between in such way that it preserves sharp edges. It could be used stand-alone to filter existing GCode(to be later used with CNC controller of your choosing), or someone could even integate it into LinuxCNC.
By the way I'm sorry for being too critical of your UI improvement proposals earlier.

taras-ko, 2016-08-22:
> Can you tell me what type of operation is it(parallel, waterline, pocket, etc.)?

It is pocketing.

> Generally to achieve precision with toolpath blending you have to set blending tolerance value far lower, like 0.02mm or even 0.01mm, and you have to generate very high quality toolpath (for example [...]

Interesting, I'll try.

> Arcs are good only for some situations, splines are ok for all the cases.
If you're interested, I can upload good code for intepolating toolpaths with Catmull-Rom Splines and with edge detection to github. It goes EXACTLY through each point in toolpath, but smooths it in between in such way that it preserves sharp edges. It could be used stand-alone to filter existing GCode(to be later used with CNC controller of your choosing), or someone could even integate it into LinuxCNC.

Interesting to look at this code and try it.

>By the way I'm sorry for being too critical of your UI improvement proposals earlier.

A good criticism will polish the idea!