vilemduha/blendercam #84: [Bug] Mess with applying the scale in "Scene" property panel

taras-ko, 2016-07-21:
Well, I need to operate default units measured in millimeters.
Default measuring is 1 meter (if I use Metric units).

So by entering number 1.0 in any digit input field means 1 meter.
But I need 1.0mm. For that I've changed the "Scale" parameter in "Scene" property panel
[image]
Great. Now I have default values measured in millimeters.
But. When I'm calculating a toolpath I'm getting some strange results.

Toolpath coordinate movements are weird
>G17 G90
S12000 M03 G00 X0 Y0 Z1000
X4414.339 Y3114.341
G01 Z-3000 F500
X3114.34 Y4414.34 F1000
X3088.796 Y4440.515
X3063.897 Y4467.305
X3039.659 Y4494.693
X3016.097 Y4522.665
X2993.223 Y4551.202

But wait I'm milling an object with dimensions 10mm x 10mm x 10mm

I suppose that right now Blendercam is hardcoded for default Scale 1.000

jeffmd, 2016-07-21:
Most of my projects are in the mm size ie 30mm x 30mm.  I just type in the value with mm ie 1.2mm.  Never had to change the scene scale.

TurBoss, 2016-07-21:
when I import models in stl from freecad to mach the scale i have to put 0.001 in scale too then gcode does not match sizes

taras-ko, 2016-07-22:
> Most of my projects are in the mm size ie 30mm x 30mm. I just type in the value with mm ie 1.2mm. Never had to change the scene scale.

Strange. My Blender's default measuring when I'm filling input field with 1.0 value is 1.0 meter.

So to set 1.0 millimeter I need to input 0.001 value.

jeffmd, 2016-07-22:
I don't input 1.0.  I actually type "1.0mm".  When units are set to Metric, Blender allows you to type in the value with the units.  If I wanted 10 millimeters I could enter "10mm" or "1cm" in the input field.  For 0.05 millimeter I could type "0.05mm" or "50um" which is 50 micrometer.

When Blender is set to Imperial I can type in "12in" for 12 inches or type in "1ft".

jeffmd, 2016-07-22:
You can also do automatic unit conversion in Blender.  If Blender is set to Metric I can type in my end mill diameter as "0.125in" and Blender converts it to 3.175mm.  I could even enter "1/8in" and Blender still converts it.  Typing in "125thou" works too.

taras-ko, 2016-07-22:
> I don't input 1.0. I actually type "1.0mm". When units are set to Metric, Blender allows you to type in the value and the units. If I wanted 10 millimeters I could enter "10mm" or "1cm" in the input field. For 0.05 millimeter I could type "0.05mm" or "50um" which is 50 micrometer.

So you are forced to append 'mm' postfix for each input action.
I'm using a numpad to input digits and as for me It's not very convenient after number input also move your hand to the center of a keyboard to input unit names. Besides all other CAD/CAM software doesn't requires to input it they have default input units.

jeffmd, 2016-07-22:
https://www.blender.org/manual/interface/controls/buttons_and_controls.html?highlight=unit#units

jeffmd, 2016-07-22:
Agreed, it is tedious if entering a lot of values.  This is where the use of unit scale comes in handy.

jeffmd, 2016-07-22:
Ok, I see what the problem is.  Stock blendercam is not taking the unit scale into consideration when exporting gcode.  Looks like I changed my blendercam in Aug 2015 to take into account the scene unit scale when calculating the unit correction value when exporting g-code.

Its a one line change in utils.py.  If you want to have this working I can post the line that needs to be added.

taras-ko, 2016-07-22:
Great! I've just remebered very cool feature where we are entering just numbers.

To move object  along any Axis by 3.5 mm I need to press:
`G, <axis name>, 3.5, <enter>`
And this is really comfortable!

jeffmd, 2016-07-22:
I was wrong.  There is only one line change in exportGcodePath() but there are a number of changes in other functions.  So not that easy to post the changes here.

jeffmd, 2016-07-22:
Looks like my changes have bugs which may be why I never submitted a pull request.

If I change the scene scale after I have switched to Blendercam mode then things go weird with measurements.  If I set the scale before switching to blendercam mode then it works.  Loading files that had a different scene scale is also an issue.  Probably why I have always stuck to a scale of 1.0.

Changing the scene scale also effects all presets that have length type properties ie they are incorrect if scene scale is not 1.0.

taras-ko, 2016-07-22:
"One Does Not Simply Fix issue #84"

taras-ko, 2016-07-22:
>  Loading files that had a different scene scale is also an issue. Probably why I have always stuck to a scale of 1.0.

Or you can first load file, than switch to Blendercam

closed 2023-01-14 by pppalain