vilemduha/blendercam #85: [Feature proposal] Extension for "medial axis" strategy.

taras-ko, 2016-07-24:
Hi.
[image]
I'm trying to engrave some letters.
As you can see engraving from the right simulation is good, but what if I want to make negative engraving (simulation on the left). There are wide surfaces which should be cutted down to flat surfaces. Is it hard to implement? Thanks.

taras-ko, 2016-07-25:
Created a workaround. It's to manually create those regions and doing pocketing on them.

ksuprynowicz, 2016-07-26:
> It's to manually create those regions and doing pocketing on them.

I think that right now it's the best option. As the pocket you drawn on the attached screenshot is in constant distance from letter contour,  you can use "Silhouette offset" function to create pocket contour. You have to enter negative value as an offset to create it inside.  The function is in the Blender CAM / Curve CAM Tools menu.

vilemnovak, 2016-07-28:
pocketing is indeed possible for medial axis strategy. I haven't added it, because I thought the sharp cutter which is used for engraving isn't good for making the surface - the lines will always be visible. would it be usefull also that way?

taras-ko, 2016-07-29:
> would it be usefull also that way?
For wood it definitely would! 
[image]
With adding an optional pocketing on medial axis strategy, you will help to make 
engravings like in attached picture.

taras-ko, 2016-07-30:
> I haven't added it, because I thought the sharp cutter which is used for engraving isn't good for making the surface - the lines will always be visible. would it be usefull also that way?

It would be useful anyway. As an option we can erase those lines with sand paper.

MadDogJroc, 2017-03-24:
Vilem, I was wondering if you'd be able to point me in the right direction on how you were thinking of implementing pocketing to the medial axis strategy.  I'd like to take a try at coding it, and have a few ideas, but am still not even close to familiar with some of the deeper workings of the project.

My issue is similar to taras-ko.  Here's an image showing the particular issue that's leading me here:
[image]

In that image, I'm trying to do a medial axis carving inside a curve (blue), using a vee bit (purple), then later doing the pocketing with an end-mill (red).  This is the fastest approach and makes the majority of the surface nice and flat/smooth.  What happens though, is there's a small protrusion left (green) where each bit misses.

I have a very specific target depth, so I can't send the medial axis operation deeper to widen it's effective diameter.  Ideally I'd be able to do a small pocket operation with the vee bit to cover only that green area that's missed.  I'm trying to speed up the operations as much as possible, so don't want to do the entire pocket area with a vee bit!

A slightly longer operation would be to do a second pass with the vee bit inside the medial axis path, which might be enough to clear things up in most situations and might be easier to code?

I can imagine various solutions, but I'm not sure what's easiest since I'm not familiar with the built-in tools that Shapely/Blender provide.

Any suggestions?

narvf, 2017-03-25:
I know this is an answer for your question but you could check F-engrave source code and  see how it's done there. I was using F-engrave for some things with great results!

MadDogJroc, 2017-03-25:
Cool, I'll check that out when I get the chance.

I spent some time trying to familiarize myself with more of the blendercam code today. (Note, I am not a seasoned Python programmer, so I'm learning the language nuances as I go!)

I'm starting to grasp some of the options we have available from Shapely, like "buffer" and "difference".  With those under my belt, I'm thinking that calculating the curve that is the green area should be relatively straight forward if we add arguments to the medial axis strategy similar to the pocket strategy (namely:  distance_between_toolpaths  and  pocket_tool_diameter.)

From there, it looks like I'd re-implement some of the pocket strategy (doesn't sound ideal to duplicate code, but I'm not sure about the re-usability of the strategy_pocket function, given that it needs an "operation" argument passed to it, which doesn't seem trivial in this case.)  It would mean taking the vee bit diameter and doing a pocket operation within the green curve using those new arguments defined in the previous paragraph.

Long term, (since I hate setting up operations) I'd combine all of this into a single new strategy, which would combine this new pocketing extension with the medial axis and traditional pocket strategies.  Then it would be possible to set up one operation to take care of all of these tool paths.  I know I'm dreaming a bit here, but it's a nice dream!

MadDogJroc, 2017-03-30:
I've managed to make a small, but functional step on this mod.  Here's some screenshots showing how it works.  I think I like it, but am still playing around with the idea.

[image]

[image]

pppalain, 2023-12-14:
separate pocket operation was added for medial

closed 2023-12-14 by pppalain