vilemduha/blendercam #86: BlenderCAM fails when trying to use Opencamlib

lpgasparotto, 2016-09-07:
Hi,
I followed the steps on the wiki and did everything mentioned on the opencamlib related issues but I can't use it. I tried with the binary download and also with the compiled blendercam add on. Also I compiled opencamlib-git with no luck, the same errors. The strange thing is the errors are different if I use the original Blender with the BlenderCAM add on and if I use the downloaded static binary.
Original Blender:
`progress{building path pattern}
progress{0.3418459892272949}
object method for retrieving silhouette
shapely getting silhouette
Self-intersection at or near point -0.0081079155206680298 0.046383921056985855
Self-intersection at or near point -0.0082680601626634598 0.058602996170520782
Self-intersection at or near point -0.008311789482831955 0.0619395412504673
Self-intersection at or near point -0.0083287768065929413 0.063235677778720856
Self-intersection at or near point -0.0082471631467342377 0.057008516043424606
Self-intersection at or near point -0.0084301959723234177 0.070973925292491913
Self-intersection at or near point -0.0080940257757902145 0.045324184000492096
Self-intersection at or near point -0.0080418903380632401 0.041346244513988495
Self-intersection at or near point -0.0081119723618030548 0.046693447977304459
Self-intersection at or near point -0.0081723760813474655 0.051302261650562286
Self-intersection at or near point -0.0081723760813474655 0.051302261650562286
Self-intersection at or near point -0.0083858966827392578 0.067593887448310852
Self-intersection at or near point 0.0081518944352865219 0.049739457666873932
Self-intersection at or near point 0.0080286785960197449 0.040338099002838135
Self-intersection at or near point 0.0080418903380632401 0.041346244513988495
Self-intersection at or near point 0.0080555379390716553 0.042387492954730988
Self-intersection at or near point 0.0082742525264620781 0.059075437486171722
computing in parts
1
joining
8.85910439491272
Polygon 1
[<shapely.geometry.polygon.Polygon object at 0x7f022b8e2828>]
Traceback (most recent call last):
  File "/home/luispa/blendercam/scripts/addons/cam/ops.py", line 219, in execute
    utils.getPath(context,o)
  File "/home/luispa/blendercam/scripts/addons/cam/utils.py", line 3347, in getPath
    getPath3axis(context,operation)
  File "/home/luispa/blendercam/scripts/addons/cam/utils.py", line 2913, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/luispa/blendercam/scripts/addons/cam/utils.py", line 340, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/luispa/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 106, in oclSample
    exportModelsToSTL(operation)
  File "/home/luispa/blendercam/scripts/addons/cam/opencamlib/opencamlib.py", line 88, in exportModelsToSTL
    bpy.ops.export_mesh.stl(check_existing=True, filepath=file_name, filter_glob="*.stl", use_selection=True, ascii=False, use_mesh_modifiers=True, axis_forward='Y', axis_up='Z', global_scale=1.0)
  File "/usr/share/blender/scripts/modules/bpy/ops.py", line 189, in __call__
    ret = op_call(self.idname_py(), None, kw)
TypeError: Converting py args to operator properties: : keyword "use_selection" unrecognized

location: <unknown location>:-1`

Static binary:
`progress{building path pattern}
progress{0.3310356140136719}
image method
progress{getting zbuffer}

time of image to numpy 0.07036805152893066
progress{0.2567923069000244}
Polygon 3
Polygon 3
Polygon 3
[<shapely.geometry.polygon.Polygon object at 0x7fe5d4bf7128>, <shapely.geometry.polygon.Polygon object at 0x7fe5dace6fd0>, <shapely.geometry.polygon.Polygon object at 0x7fe5d4bf70f0>]
Traceback (most recent call last):
  File "/home/luispa/Descargas/blender-2.76b-linux-glibc211-x86_64_CAM_0.9/2.76/scripts/addons/cam/opencamlib/oclSample.py", line 6, in <module>
    stl = camvtk.STLSurf(tempfile.gettempdir()+"/model0.stl")
  File "/usr/lib/python2.7/dist-packages/camvtk.py", line 725, in __init__
    self.mapper.SetInput(self.src.GetOutput())
AttributeError: SetInput
Traceback (most recent call last):
  File "/home/luispa/Descargas/blender-2.76b-linux-glibc211-x86_64_CAM_0.9/2.76/scripts/addons/cam/ops.py", line 214, in execute
    utils.getPath(context,o)
  File "/home/luispa/Descargas/blender-2.76b-linux-glibc211-x86_64_CAM_0.9/2.76/scripts/addons/cam/utils.py", line 3233, in getPath
    getPath3axis(context,operation)
  File "/home/luispa/Descargas/blender-2.76b-linux-glibc211-x86_64_CAM_0.9/2.76/scripts/addons/cam/utils.py", line 2591, in getPath3axis
    chunks.extend(sampleChunks(o,pathSamples,layers))
  File "/home/luispa/Descargas/blender-2.76b-linux-glibc211-x86_64_CAM_0.9/2.76/scripts/addons/cam/utils.py", line 327, in sampleChunks
    oclSample(o, pathSamples)
  File "/home/luispa/Descargas/blender-2.76b-linux-glibc211-x86_64_CAM_0.9/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 111, in oclSample
    chunkPointSamplesFromCSV(chunks)
  File "/home/luispa/Descargas/blender-2.76b-linux-glibc211-x86_64_CAM_0.9/2.76/scripts/addons/cam/opencamlib/opencamlib.py", line 48, in chunkPointSamplesFromCSV
    csv_file = open(tempfile.gettempdir()+'/ocl_chunk_samples.txt','r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ocl_chunk_samples.txt'

location: <unknown location>:-1

location: <unknown location>:-1`

I'm using KDE Neon which is based on Ubuntu 16.04

Please tell me if you need more information.

ksuprynowicz, 2016-09-10:
Hi!
Can you try with Blender 2.77a? Between 2.76 and 2.77a Blender STL export command was changed, and I think that the problem is related to that.
Let me know it it works.

lpgasparotto, 2016-09-11:
Thank you for your reply! Should I compile it from the source? I guess there's not a 2.77a binary package in the repos.

ksuprynowicz, 2016-09-11:
> Should I compile it from the source? I guess there's not a 2.77a binary package in the repos.

Sure, download dependencies and give it a try!
To make it easier apt package manager has 'build-dep' option to automatically download dependencies:

    apt build-dep blender

taras-ko, 2016-09-12:
You may download already compilled version from blender web site.

lpgasparotto, 2016-09-15:
Is there a 2.77a binary package in the website? I can't find it.

ksuprynowicz, 2016-09-18:
> Is there a 2.77a binary package in the website? I can't find it.

There isn't. The newest Blender+BlenderCAM packages are 2.76b. If you don't have 2.77 in Ubuntu repositorym then easiest way is building it from source.

taras-ko, 2016-09-18:
Afaik. Blendercam is a python module, you don't need to build it. Just download the newest version of Blender from blender.org and clone repository with Blendercam.

ksuprynowicz, 2016-09-19:
> Afaik. Blendercam is a python module, you don't need to build it. Just download the newest version of Blender from blender.org and clone repository with Blendercam.

You're right, but Blender downloaded from repository has it's own python distribution bundled with it, and I think that it doesn't have required dependencies: python3-shapely and python3-numpy.
As python distribution bundled with Blender is not the same as the system-wide one, dependencies installed form apt will not be available in Blender. Maybe there's a way to add them to bundled python, but I have no idea how(maybe smoething like pythons package mangager - PIP).

https://github.com/vilemnovak/blendercam/wiki/Using-Blendercam-from-github
https://github.com/vilemnovak/blendercam/wiki/Using-BlenderCAM-with-OpenCAMLib

pppalain, 2023-01-14:
old version of blender

closed 2023-01-14 by pppalain